ecocurious2/MultiGeiger #159: split into multiple code files

ThomasWaldmann, 2020-01-16:
currently, almost all code is in multigeiger.ino.

we could split e.g. into:
- setup / loop / isr / geiger counting part (== stays in multigeiger.ino)
- transferring stuff to LoRa / internet based services (transfer.ino?)
- log output in misc. formats (logging.ino?)

t-pi, 2020-01-20:
Please consider also planned Bluetooth transfer of the data to a connected Smartphone in the communication module.

ThomasWaldmann, 2020-02-11:
I would like to start on this soon, but this needs a bit of coordination as likely a lot of code will move and might lead to merge issues.

So, @rexfue please notify me when it is a good time to do that.

The good thing will be that after that, we'll maybe have a bit less conflicts (at least if we work in different files) and easier working because the files won't be that huge.

rexfue, 2020-02-11:
OK, I hope I get #189 and #190 done tomorrow. And then you can rearrange the code. Afterwards I will then work on #140 again.
I inform you, when #189/190 are done (you'll see it on the next PR anyway)

ThomasWaldmann, 2020-02-16:
mostly done now.

didn't touch the lora and the data transmission stuff yet.

ThomasWaldmann, 2020-02-21:
done & merged.

closed 2020-02-21 by ThomasWaldmann