ecocurious2/MultiGeiger #163: create sphinx docs / readthedocs.io infrastructure

ThomasWaldmann, 2020-01-18:
Related to #52, but at first rather the "infrastructure part", setting everything up.

ThomasWaldmann, 2020-01-18:
@rexfue I can't setup the readthedocs.io documentation publishing site because I do not have enough permissions here, can you upgrade me? Need to do "Settings" here.

ThomasWaldmann, 2020-01-18:
Also it would be good if you @rexfue and @WwwBoehriDe could create accounts on readthedocs.org (and tell me your username), so I can add you as additional admins there (for better "bus factor"). :-)

rexfue, 2020-01-18:
Sorry, I also don't have the rights. Thats Nicola. I tried to login as her, but there's 2FA, and she doesn't read her mails at the moment :-(     I'll try it later.

rexfue, 2020-01-18:
Username at readthedocs is **rexfue**

rexfue, 2020-01-18:
If I understand the docs correctly, only the owner has the 'settings' rights and there is only one owner, that is Nicola. See from the docs:
"There is only one owner of a repository owned by a user account; this permission cannot be shared with another user account. "

ThomasWaldmann, 2020-01-18:
Best is to create some "Core" team with us and give them "Admin" permissions.

https://github.com/ecocurious/MultiGeiger/settings/access

rexfue, 2020-01-18:
I get 404 on that link

ThomasWaldmann, 2020-01-18:
Yes, because you do not have the permissions to do that (I get the same, but the corresponding link for a repo I own works).

rexfue, 2020-01-18:
OK.
I’m tomorrow not at home before late evening. Maybe I can contact Nicola anyway.
Or you try to contact her and ask her do that.

ThomasWaldmann, 2020-01-18:
@ecocurious Hi, kannst Du das bitte machen? (siehe oben)

Bitte den Github-Issue via Web-Browser lesen, nicht nur die E-Mail - Link steht in der E-Mail.

ecocurious, 2020-01-18:
Hallo ihr lieben, ich bin jetzt ein paar tage offline, am 24. kann ichs machen

ThomasWaldmann, 2020-01-18:
https://multigeiger.readthedocs.io/en/latest/ (this is generated from my repo / docs-structure PR branch until the setup at ecocurious org has been made).

ThomasWaldmann, 2020-01-24:
@ecocurious und, wieder online? siehe oben.

ecocurious, 2020-01-25:
habe reinhard den gewünschten code schon gemailt

ThomasWaldmann, 2020-02-01:
Any news here?

I still don't have permissions, so someone else needs to set this up or give me permissions.

rexfue, 2020-02-02:
Back on multigeiger now  :-)
I looked in settings of my own repos, but didn't find something to 'create a Core Team' as you mentioned a few comments above. So please tell me the way, what exactly to do, then I will try.
Or call Nicola to get her account credentials and log in with them and do it yourself.

ThomasWaldmann, 2020-02-03:
Best is to create some "Core" team with us and give them "Admin" permissions.

- go to https://github.com/ecocurious/MultiGeiger/settings/access (may ask for password there)

- Teams -> "+ Create new team"

- "core team", "core developers", (no parent team), [X] visible, [Create Team]

- on the page you see next, add all core developers as team members

- go back to https://github.com/ecocurious/MultiGeiger/settings/access

- Teams -> "Add a team", choose "core team"

- "core team" is now in the list of teams, to the right of the team name, choose "admin" permissions from the dropdown menu.

ThomasWaldmann, 2020-02-03:
@rexfue @ecocurious see the above comment (on github).

rexfue, 2020-02-03:
OK, seen. I try it, if I can contact Nicola.

ThomasWaldmann, 2020-02-09:
OK, permissions fixed now, readthedocs set up. Closing.

closed 2020-02-09 by ThomasWaldmann