ecocurious2/MultiGeiger #168: english / german docs on readthedocs.io

ThomasWaldmann, 2020-01-18:
There seems to be a way how to deal with multi-language content on readthedocs.io, but I never tried it.

Guess one needs to reflect that somehow in the repo dir / file structure also.

Find out how.

ThomasWaldmann, 2020-02-09:
https://docs.readthedocs.io/en/stable/guides/manage-translations.html

I never did multi-language documentations (I always just used english, because all admins / developers / nerds understand it anyway and not having to translate docs saves quite some work).

But if the target audience are "normal people", translations might be required.

I did several approaches in the past for translations and using `transifex` worked best, because translators often know that platform and it has a comfortable UI to translate / manage translations.

The "original language" usually should be English btw, because there are more translators for `English to X` than e.g. for `German to X`.

rexfue, 2020-02-09:
But I guess it  is easier and much faster for all of us to write docs in german.
You can simply translate  the german text with deepl.com.

ThomasWaldmann, 2020-02-09:
Just imagine you find a nice project and everything is in chinese (or portuguese) - would you use it?

Being in English is IMHO one of the main criteria for a project to be widely (== internationally) usable and successful.

And atomic power plants are (almost) everywhere, so everywhere the same problem.

t-pi, 2020-05-19:
There's a new channel on sensor.community, and they already stumbled over the german docs...
https://chat.luftdaten.info/airquality/channels/radiation
@WwwBoehriDe maybe in that channel there'd be people willing to help with an english version of the instructions for assembly and setup?

ThomasWaldmann, 2020-05-21:
I'm setting up translated docs now, using transifex (which is quite comfortable even for non-programmers) and readthedocs.io (which supports translated docs).

closed 2020-05-21 by ThomasWaldmann

ThomasWaldmann, 2020-05-22:
- https://www.transifex.com/thomaswaldmann/multigeiger  <-- deutsche uebersetzung kann man dort ergänzen.
- https://multigeiger.readthedocs.io/de/latest/
- https://multigeiger.readthedocs.io/en/latest/

PR #320 (2020-05-21): translated docs
fixes #168.