ecocurious2/MultiGeiger #170: silence?

ThomasWaldmann, 2020-01-18:
I have the MultiGeiger running since a while and usually it is ticking normally.

But sometimes, there are multiple seconds silence, not a single tick. Why? I have a source with ~250 CPM close by, so no GM event for multiple seconds is highly unlikely.

Not sure if it is related, but I noticed this since flashing a binary (V1.12.0) that has sending to madwifi and sensor.community enabled (but I did not register / configure anything special related to that). Is there some buffer filling up and then blocking somehow?

rexfue, 2020-01-19:
Yes, this seems to be a timeout in the http library.  But this does not affect the count ( interrupt !! ).

t-pi, 2020-01-20:
Indeed, I noted the same tonight (2020-01-19 evening). Madavi was not responding and the Multigeiger seemed stuck while "Sending to Madavi...". But if the counts are still registered, guess it's not critical...

ThomasWaldmann, 2020-05-08:
To be more precise:

The root cause for the intervals of silence is that the ticking in current firmware is done in main loop.

If the main loop is stuck in some data transmission call (and waits until some timeout in there), it won't call the code doing the speaker/led tick. PR #251 is ongoing work on this, doing the tick sound via PWM (works already) and later also turning tick sound on from GM pulse ISR and off by a timer ISR (TODO).

The HV charging is also done from within main loop (will be changed by PR #254 to be timer interrupt driven), but at low radiation levels a few seconds with no hv charge is no problem AFAIK.

As @rexfue noted, the GM pulse counting is already interrupt driven, so counts won't be missed.

ThomasWaldmann, 2020-05-12:
Should be solved by the mentioned PRs, see previous comment.

closed 2020-05-12 by ThomasWaldmann