ecocurious2/MultiGeiger #174: Porting to Olimex ESP32-POE

hagenbuch, 2020-01-22:
Hi all,

great device, great software, thanks very much!

However, where I'd like to use the device, I will have no Wifi but Ethernet and POE (if you have to run a cable anyway, why send more than one?) so I bought a Olimex ESP32-POE

https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware

(Yes there is a more expensive version with a better ETH transformer but I guess IM fine with the "old" one, I have revision B).

I managed to flash the simple ethernet demo 

https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino

on the Olimex but I'm not sure where to start to merge their code with yours. Does someone have a starting point?

Would be awesome. I think cabling can be made by dupont cables.. I also would like to read from an BME280 at the same time.

rexfue, 2020-01-23:
I think if you use the same PINs as our board (also for the I2C bus), our software should work 'out of the box' with WLAN.  An adaptation to ethernet should then be quite easy. Since I don't have such hardware, I can't try that.
Please keep the connection with Dupont cables as short as possible - the inputs are very high impedance and crosstalk is easy. 
Btw. BME280 is included in the software.

ThomasWaldmann, 2020-02-21:
The (wireless)LAN related code is now in `webconf.*`.

`transmission.*` expects a working network connection (but does not care whether it is wireless or wired or LoRa).

ThomasWaldmann, 2020-05-11:
As there was no activity in here, guess we can close this.

closed 2020-05-11 by ThomasWaldmann