ecocurious2/MultiGeiger #177: style cleanup

ThomasWaldmann, 2020-02-03:
the code would be better readable and feel more consistent with cleanups like:

- use "a, b" - not "a,b"
- use "a <operator> b" - not "a<operator> b" nor "a <operator>b" nor "a<operator>b"

is there some C style checker we could adapt and integrate via travis-ci?

ThomasWaldmann, 2020-02-09:
@rexfue I could do this, but considering the amount of issue that are currently assigned to you, I suspect you have a lot of local, not-yet-published code changes and I'ld like to avoid merge conflicts.

So just notify me when it is a good time for a code style cleanup PR.

rexfue, 2020-02-09:
OK. At the moment I'm working on the web interface to show/set config and show actual measured values. This will be a pretty big change but i put almost everything in an extra file. So i guess, it would be fine, to do the cleanup now. I can put in my changes after that (maybe by hand). But for the next two weeks or so, I cant't get around to it, so feel free to do the cleanup.

ThomasWaldmann, 2020-02-09:
Ah, ok, great you kept it in a separate file, so we have less to do for the modularizing ticket.

ThomasWaldmann, 2020-02-09:
Working on this now...

ThomasWaldmann, 2020-02-10:
I found a good-working automated code formatter. Way better than manually fixing this.

PR #183 (2020-02-11): use astyle code formatter for consistency and style
astyle is a configurable automatic C/C++ code formatter.

i did the cleanup step-by-step:
- first config change
- then automatic formatting
- then manually fix stuff that didn't look nice
- repeat

fixes #177.
closed 2020-02-11 by ThomasWaldmann