ecocurious2/MultiGeiger #180: Wifi abschalten und Angleichung Lora

sbiermann, 2020-02-07:
Hallo,
gibt es die Möglichkeit das Wifi komplett abzuschalten? Ich nutze Lora und daher brauche ich kein aktives WLAN. 
Mir ist bei Lora noch aufgefallen das dort nicht alle gesammelten Daten übertragen werden. Bei der Übertragung an sensor.community werden wesentlich mehr Werte übertragen als bei Lora. Ist geplant dies anzugleichen? Ich würde dann nämlich mittels Node-Red die Daten von TTN an sensor.community weiter leiten, aber es scheinen da noch ein paar Werte zu fehlen damit das funktioniert.

Viele Grüße
Stefan

rexfue, 2020-02-07:
Hi Stefan,
WLAN abschalten ist z.Zt noch nicht vorgesehen.Kommt vielleicht in einer späteren Release.
Was meinst Du mit 'nicht alle gesammelten Daten'? Unser Geigerzähler 'sammelt' bzw. zählt die Impulse des Zählrohres und dieser Wert wird z.Zt. per LoRa übertragen.

Allerdings ändern wir gerade die Daten, die per LoRa übertragen werden um eine einfachere Anbindung an sensor.community zu erreichen. Beschreibung der Payload siehe https://github.com/ecocurious/MultiGeiger/tree/master/docs/ttn2luft.pdf
Über die HTTP-Integration von https://ttn2luft.citysensor.de wird dieses Format dann umgesetzt und  nach sensor.community gesendet.

sbiermann, 2020-02-07:
Den Unterschied den ich meine ist das hier:
`  String body = "{\"software_version\":\""+String(revString)+"\",\"sensordatavalues\":[";
  body += "{\"value_type\":\""+valuetype+"\",\"value\":\""+current_cpm+"\"}";
  body += ",{\"value_type\":\"hv_pulses\",\"value\":\""+String(hvpulses)+"\"}";
  body += ",{\"value_type\":\"counts\",\"value\":\""+String(GMC_counts_2send)+"\"}";
  body += ",{\"value_type\":\"sample_time_ms\",\"value\":\""+String(timediff)+"\"}";
  body += "]}";` wenn es um die Daten sensor.community geht. Während bei Lora nur:
`// first the number of counts
    ttnData[0] = (GMC_counts_2send >>24) & 0xFF;
    ttnData[1] = (GMC_counts_2send >>16) & 0xFF;
    ttnData[2] = (GMC_counts_2send >>8) & 0xFF;
    ttnData[3] = GMC_counts_2send & 0xFF;
    // now 3 bytes for the time in ms for this numer of counts (max ca. 4 hours)
    ttnData[4] = (timediff >> 16) & 0xFF;
    ttnData[5] = (timediff >> 8) & 0xFF;
    ttnData[6] = timediff & 0xFF;      
             `
sind. Der größte Unterschied der mir aufgefallen ist, ist das hvpulses, das dieses fehlt. Daher war ich der Meinung es sei ein wichtiger Bestandteil der Datenlieferung an sensor.community welcher noch nicht bei Lora enthalten ist.

Gibt es eine Doku zu https://ttn2luft.citysensor.de, wie man die Daten in welchen Format hinschicken muss?

rexfue, 2020-02-07:
OK, das hv_pulses ist eine Debug-Größe. Da es bei LoRa auf jedes Byte ankommt, übertragen wir das darüber nicht.

Daten zu ttn2luft: da braucht man nix machen, nur bei TTN die HTTP-Integration auf diese Adresse stellen und KEINEN Payload-Decoder verwenden. Dann sollte das klappen.
Bei sensor.community ist dieser Sensor dann mit TTN-xxxxx anzumelden, wobei xxxxx die letzten 4 Byte der DEVEUI in dezimal sind

sbiermann, 2020-02-07:
Ok, verstehe, ich habe den Sensor bei sensor.community registriert und dann so wie du geschrieben hast per HTTP Integration in der TTN Console eingerichtet. Die ersten Daten sind auch angekommen. Er taucht zwar noch nicht auf der https://multigeiger.citysensor.de/ Karte auf, aber das ist vermutlich nur eine Frage der Zeit bis die Daten von einem System ins andere gehen.

rexfue, 2020-02-07:
Super. Magst Du mir noch die Sensornummer (die, die Du bei sensor.community erhalten hast) sagen? Dann kann ich auch nachschauen, ob die Daten bei mir ankommen.

sbiermann, 2020-02-07:
Die Sensor ID lautet: 41023.

rexfue, 2020-02-07:
Danke - es funktioniert

ThomasWaldmann, 2020-02-21:
WiFi abschalten waere ggf. ne sinnvolle Verwendung fuer Dip-Switch SW3 ("unused"):
- WiFi aus macht v.a. mit LoRa Sinn (die meiste Zeit)
- man will aber ggf. auch auf LoRa Multigeiger per WiFi auf die Config-Oberflaeche drauf (muss also WiFi einschalten koennen): Config und FW-Updates via Web-Oberflaeche.
- bei einem MultiGeiger-Mobilgerät schaut man auch eher auf's Display als dass man (die meiste Zeit) WiFi an haben muesste. Ab und zu aber doch, s.o.

ThomasWaldmann, 2020-05-11:
https://github.com/espressif/arduino-esp32/issues/1077

ThomasWaldmann, 2020-05-11:
BTW, is there anything left to do for LoRa? Never put 2 different topics into 1 github issue...

rexfue, 2020-05-12:
No, nothing to do for LoRa here.

ThomasWaldmann, 2020-05-13:
See #289.

closed 2020-05-13 by ThomasWaldmann