ecocurious2/MultiGeiger #184: Use TLS (https) for API access?

justinotherguy, 2020-02-10:
Moin,

If I'm not mistaken, currently we're using (unencrypted) http only to access the APIs:
https://github.com/ecocurious2/MultiGeiger/blob/master/multigeiger/multigeiger.ino#L198

What's the reason for this? The ESP32 should have enough power - do the libraries we use support TLS? Have we done a test for this yet?

Regards, J.

ThomasWaldmann, 2020-02-10:
Yeah, that's the current state.

https would be better of course, but needs some infrastructure / environment:
- https on server side
- precise time on client side (ntp)
- https on client side (space requirements = ?)
- actually checking the server's authenticity (because I guess we mostly are after that, not primarily after encrypting data which will be published anyway)

justinotherguy, 2020-02-10:
- https on server side
-> a quick test showed that all 3 currently used APIs also provide https; and the cert setup looks proper, too -> that should not be a problem, so

- precise time on client side (ntp)
-> ok - we do need a source for time to assess the validity of the certificates, true; we should be able to obain that via ntp; but - that needs to be implemented first, too - correct

- https on client side (space requirements = ?)
-> true; I had only thought of CPU power - but we also need RAM for the code

- actually checking the server's authenticity (because I guess we mostly are after that, not primarily after encrypting data which will be published anyway)
-> true; checking data integrity or authenticity of the client doesn't make sense in this setup, either :)

ThomasWaldmann, 2020-02-10:
Well, checking authenticity of the client and data integrity actually would make sense, IMHO.

But I do not know whether / how authenticity check is done by these APIs (username / password maybe?).

Data integrity would be only done on the TCP level currently, which is ok against accidental corruption, but not against attacks.

ThomasWaldmann, 2020-02-10:
And if it is done by username/password, that is of course an additional reason to use https.

ThomasWaldmann, 2020-02-10:
https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino

https://github.com/espressif/arduino-esp32/blob/master/libraries/HTTPClient/examples/HTTPClientEnterprise/HTTPClientEnterprise.ino

Note: HTTPCLIENT_1_1_COMPATIBLE api functions are deprecated.

ThomasWaldmann, 2020-02-10:
# MADAVI "http://api-rrd.madavi.de/data.php"

# SENSORCOMMUNITY "http://api.sensor.community/v1/push-sensor-data/"

# TOILET "http://ptsv2.com/t/enbwck3/post"

ThomasWaldmann, 2020-02-10:
Found some timing data on the web and they say a https (tls) connect takes ~2s.

So, we could optimize the client so it tries to keep the connection open, putting additional burden on the server (lots of connections if there are lots of geigers). Also, the server might still disconnect.

Or we don't and wait 2s...

ThomasWaldmann, 2020-02-10:
BTW, we maybe should avoid loading firmware updates over https / tls - otherwise we could run into a dead lock situation:
- assuming we need to update the CA cert (because it expired or CA change or ...).
- if the update is fetched via https (tls) and we can't connect due to a cert issue, we can't update the CA cert that is in the firmware either.

So rather fetch firmware via http and use a signed firmware image, see #120.

ThomasWaldmann, 2020-02-10:
Debugging TLS: on Arduino IDE, set core debug level to verbose.

ThomasWaldmann, 2020-02-11:
btw, it doesn't look like the arduino-esp32 lib checks for certificate expiry. at least it had no problem with the certs when i set the esp32 time to y2038.

ThomasWaldmann, 2020-02-11:
another issue is that the CA certs aren't valid for a very long time.
so, if we include them into our firmware we must not forget to update them 2021.

how the cert rollover should work best is not totally clear to me yet.

ThomasWaldmann, 2020-02-11:
[https_test.txt](https://github.com/ecocurious2/MultiGeiger/files/4183635/https_test.txt)

ThomasWaldmann, 2020-02-16:
https://letsencrypt.org/certificates/

http://esp32-server.de/wificlientsecure/

https://esp32.com/viewtopic.php?t=7585#p31952
https://www.esp32.com/viewtopic.php?t=3870#p17576

https://www.sslshopper.com/certificate-decoder.html

ThomasWaldmann, 2020-02-16:
So the solution might be to concatenate all root CAs we need (misc CAs, old certs, new certs, ...).

And really use the root certs, not the letsencrypt X3 intermediate CA cert.

ThomasWaldmann, 2020-02-16:
[https_test.txt](https://github.com/ecocurious2/MultiGeiger/files/4211018/https_test.txt)

Looks like that works! \o/

justinotherguy, 2020-02-16:
excellent - sounds like a good approach \o/

ThomasWaldmann, 2020-02-21:
TODO: update release making checklist:

Check whether some CA certificate will expire soon and whether we already can add their next valid cert.

ThomasWaldmann, 2020-04-14:
there's some code in #236 to review and practically test.

PR #236 (2020-05-08): TLS support
- added NTP support (for TLS, we should have correct date/time)
- added persistent per-server connection instances
- use keep-alive for connections
- add the required CA certs for sensor community, madavi, toilet
- use TLS (https) for data transmission

I did not practically test it yet, but reused some tested code.

Fixes #184.
closed 2020-05-08 by ThomasWaldmann