ecocurious2/MultiGeiger #190: platformio compile issues seen on travis-ci

ThomasWaldmann, 2020-02-11:
```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/heltec_wifi_kit_32.html

PLATFORM: Espressif 32 1.11.1 > Heltec WiFi Kit 32

HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash

DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)

PACKAGES: toolchain-xtensa32 2.50200.80 (5.2.0), framework-arduinoespressif32 2.10004.191002 (1.0.4), tool-esptoolpy 1.20600.0 (2.6.0)

Converting multigeiger.ino

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:135:0: warning: "PIN_SWI_0" redefined

 #define PIN_SWI_0 39

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:130:0: note: this is the location of the previous definition

 #define PIN_SWI_0 36

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:136:0: warning: "PIN_SWI_1" redefined

 #define PIN_SWI_1 38

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:131:0: note: this is the location of the previous definition

 #define PIN_SWI_1 37

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:137:0: warning: "PIN_SWI_2" redefined

 #define PIN_SWI_2 37

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:132:0: note: this is the location of the previous definition

 #define PIN_SWI_2 38

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:138:0: warning: "PIN_SWI_3" redefined

 #define PIN_SWI_3 36

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:133:0: note: this is the location of the previous definition

 #define PIN_SWI_3 39

 ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:933:66: warning: missing terminating " character

   String body = "{\"software_version\":\"" + String(revString) + "\",\"sensordatavalues\":[\

                                                                  ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:934:79: warning: missing terminating " character

 {\"value_type\":\"" + temp + "\",\"value\":\"" + String(bme_temperature, 2) + "\"},\

                                                                               ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:935:76: warning: missing terminating " character

 {\"value_type\":\"" + humi + "\",\"value\":\"" + String(bme_humidity, 2) + "\"},\

                                                                            ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:936:77: warning: missing terminating " character

 {\"value_type\":\"" + press + "\",\"value\":\"" + String(bme_pressure, 2) + "\"}\

                                                                             ^

/home/travis/build/ecocurious2/MultiGeiger/multigeiger/multigeiger.ino:937:3: warning: missing terminating " character

 ]}";
```

ThomasWaldmann, 2020-02-11:
The #define issues seem false positives as the originals and duplicates are in a #if / #else. 

The string assembly looks a bit ugly anyway, should be solved in another way that needs less escaping and + operators, sprintf maybe?

rexfue, 2020-02-11:
I work on this at the moment - I will pack it into a function

ThomasWaldmann, 2020-02-18:
Seems like #200 fixed that.

closed 2020-02-21 by ThomasWaldmann