ecocurious2/MultiGeiger #198: how to flash a firmware binary with esptool.py

ThomasWaldmann, 2020-02-13:
we have no docs about this, right? but we should have some, because normal users do not want to install the full arduino ide and compile from source / upload to the device via ide.

ThomasWaldmann, 2020-02-13:
First, erase the flash - usually something like:
``` 
esptool.py --port /dev/ttyUSB0 --baud 115200 --chip esp32 erase_flash
```

Then upload the binary (this commandline is intercepted from the arduino IDE shell call):
```
esptool.py --chip esp33 --port /dev/ttyUSB0 --baud 115200 \
  --before default_reset --after hard_reset \
  write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect \
  0xe000 .arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/boot_app0.bin \
  0x1000 .arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/bin/bootloader_qio_40m.bin \
  0x10000 multigeiger.ino.bin \
  0x8000 multigeiger.ino.partitions.bin
```

Is that also possible in a simpler way? Also, is that really **esp33**?

rexfue, 2020-02-14:
Why do we need this? We have OTA !

Sure, it must be --chip esp32. And if you have an already programmed chip (as usual), you MUST NOT erase, you only have to flash the program. Then the line is shorter:

esptool --chip esp32 --port COM4 --baud 921600 --before default_reset 
--after hard_reset write_flash -z --flash_mode dio --flash_freq 40m 
--flash_size detect 0x10000 multigeiger.ino.bin

ThomasWaldmann, 2020-02-14:
Well, we developers flash something all the time, but normal users may not have an OTA capable version yet on their device.

Also, guess it somehow needs to be partitioned for OTA once and after that I guess one must use OTA (not sure how one deals with the 2 flashable partitions via esptool).

M-a-r-i-a-n-n-e, 2021-03-21:
I suggest to also mention this solution on each release page where you can download the precompiled bin file for 0x1000 (maybe provide the other bins there, too) and also point to the esptool and the [Flash Download Tool](https://www.espressif.com/en/support/download/other-tools) on this page.

Morg42, 2024-12-26:
After forgetting all login data and prematurely erasing the flash, I'm trying to flash the ready-made bin to the device, I can't get the current IDE to compile multigeiger for me :-/

Any hints how to "just flash, don't compile" properly? I don't have a multigeiger.ino.partitions.bin..