ecocurious2/MultiGeiger #206: is arduino-lorawan useful for us?

ThomasWaldmann, 2020-02-17:
https://github.com/mcci-catena/arduino-lorawan

made by same guy as the lorawan-lmic we already use.

fafu, 2020-02-17:
It seems a good approach to separate lora related code from the rest of the code. For LoRa, the current implementation is blocking until the network could be joined - I even had several hangs when there are network related issues (only on LoRa). I locally solved this issue by using code from the Paxcounter project with dedicated send and receive threads and messages queues - it seems to work fine. The proposed library offers a similar approach in a comfortable way.

ThomasWaldmann, 2020-04-13:
does somebody know if using this library would "fix" the currently required manual editing of library code (see our github README).

ThomasWaldmann, 2021-03-25:
not needed for now.

closed 2021-03-25 by ThomasWaldmann