ecocurious2/MultiGeiger #214: Set dummy TOILET address

rexfue, 2020-02-21:
Because this address is individual. Everyone should  request their own address to send data to.
This could be explained with comments.

ThomasWaldmann, 2020-04-13:
This could also be in the web-based configuration form maybe?

Or in `userdefines.h`?

rexfue, 2020-04-13:
I think, we should leave this completely out. It was only useful for me during testing. The normal user don't need it.

ThomasWaldmann, 2020-04-13:
I might need this for testing tls support, too.

jensliebau, 2020-04-16:
I would vote for not removing it, but enhance it:

If you take a look at "Feinstaubsensor" https://luftdaten.info/feinstaubsensor-bauen they provide support for two "private" API Endpoints.
* "An eigene API senden"
* "Senden an InfluxDB"

In the long run it would be great to add support for both in mulitgeiger, too.

ThomasWaldmann, 2020-05-03:
Related to #115?

ThomasWaldmann, 2020-05-11:
I am adding some comment for it now (see top post).

ThomasWaldmann, 2020-05-11:
@jensliebau I kept it simple in PR #272 (2nd commit) to avoid feature creep in there.

For more, please open a fresh ticket.

commit 9a335a57 (2020-05-13): toilet -> custom server, add comments about toilet usage, fixes #214
closed 2020-05-13 by ThomasWaldmann