ecocurious2/MultiGeiger #220: simplify / speed up the ISR, reduce locked time

ThomasWaldmann, 2020-02-22:
I'ld like to:
- simplify, reduce and speed up the ISR code
- simplify the code in the main loop interacting with the ISR variables
- reduce locked time by less and shorter critical sections
- do all pulse count bookkeeping in the main loop

What I think of is this:
- keep the ISR code as minimal as possible, especially remove isr_GMC_counts_2send, isr_count_timestamp_2send and isr_gotGMCpulse.
- all the different counters for display, 1-minute-log, 2.5-minute-transmission, all-time, ... would be just kept in the main loop and updated there (not in ISR).
- enclose PIN_TEST_OUTPUT operations in some #ifdef TEST_PIN_ENABLE .. #endif

rexfue, 2020-02-23:
Fine.
Then please add 1 sec timer for hv pulses (see #192).

ThomasWaldmann, 2020-02-23:
That's a different issue, not in scope of this issue.

PR #247 (2020-05-04): simplify ISR, refactor main loop
fixes #220 and #66.  #38 can now trivially be addressed also.

this is a rather big and likely hard to review change, but the code needed this as the lots of bookkeeping variables and the rather lengthy main loop and strange coupling in there was not easily manageable / understandable any more. maybe it is easier to look at the resulting code rather than at the diff.

guess the ISR is now as simple and as fast as it gets, no duplication and no longer-term bookkeeping in there any more. also no debugging overhead when not needed.

the main code does not interact [... cut]
commit 7136aa3b (2020-05-04): simplify ISR, bookkeeping in main loop, fixes #220
closed 2020-05-04 by ThomasWaldmann