ecocurious2/MultiGeiger #230: compile-time vs. runtime hw selection

ThomasWaldmann, 2020-03-26:
currently, some hw selection happens via `#define` in userdefines.h.

this requires compiling / publishing one binary per different hardware (for us) and selecting the right binary to flash (for the user). also, some compile time issues might only get noticed for a specific selection of hw / sw features.

can we do the selection at runtime (like we already have for the THP sensor) and build a universal binary for all hardwares? either automated detection/selection or manual enabling of hw features via the web configuration page?

rexfue, 2020-03-27:
This is a good - because old - idea. I had this in mind from the very beginning. I wanted to do it with a PIN. But there was no one available.
I will do a test next week, if it works for example by querying the LoRa chip.

ThomasWaldmann, 2020-04-08:
fixed by #231 ?

rexfue, 2020-04-08:
I think so

closed 2020-04-08 by ThomasWaldmann