ecocurious2/MultiGeiger #232: Updated to

jensliebau, 2020-04-06:
Hi,

I've build a multigeiger based on "Heltec WiFi Kit 32" when I've received it, it has had version 1.12.0 installed. 
* It connects to my Wifi without problem

To test the new webconfig and to disable the speakerTick I've updated form 1.12.0 to 1.12.1-dev (
https://github.com/ecocurious2/MultiGeiger/commit/8223a2840d1df9cbd830bc38ece43d5f40e5de23)

* Unfortuntely my mulitigeiger no longer connects to my Wifi
* Display shows "Connecting..." but it does not
* It does not fall back to AP-mode

rexfue, 2020-04-06:
Are you sure, that it doesn't connect?  I guess, there is no connection problem but a display problem: the 'connecting...' string will not be cleared in this version. 
Can you prove the connection with your router or at madavi.de?

jensliebau, 2020-04-06:
> Are you sure, that it doesn't connect?

Yes, I'm sure. It does not show up as a Wifi-Client in my Wifi-Dashboard.

rexfue, 2020-04-06:
OK. Try to connect to its own Accesspoint direct after reset and renew your WiFi credentials via the web page.

jensliebau, 2020-04-06:
I'did some more research and recompiled 1.12.1-dev with DEBUG serial output. It did show:

> Config size: 177
> Wrong config version.

As suggested I've reconfigured WiFi credentials while multigeiger was in AP-mode. It's working correct again. 👍 

But maybe there are some config problems when upgrading to the new version, which should be checked?

rexfue, 2020-04-07:
OK, fine that it works now.
The 1.12.1-dev is work in progress, as the 'dev' suffix says. We had to change the layout of the config page. You now can set the variables for speaker tick, start sound etc. per web ui.
All this will be in the next official release (1.13.0), which we try to have ready by the end of the month. Then also the readme will be updated.

ThomasWaldmann, 2020-04-07:
Guess the question here rather is whether we can handle this more gracefully / automatically.

jensliebau, 2020-04-07:
> Guess the question here rather is whether we can handle this more gracefully / automatically

yes, this was one of the the reason of my report.

Don't worry, I knew this is a dev version and I don't expect it to work without problems. Most often release quality software didn't work without problems, too. ;-) 

Nevertheless most of the times it might be better to report problems.

rexfue, 2020-04-08:
OK, thanks for the report.

ThomasWaldmann, 2020-04-13:
PR #233 fixes the docs for this.

closed 2020-04-14 by ThomasWaldmann