ecocurious2/MultiGeiger #234: loraWan.cpp: another compiler warning

ThomasWaldmann, 2020-04-13:
```
sketch/loraWan.cpp:60:3: warning: #warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!" [-Wcpp]
 # warning "You must replace the values marked FILLMEIN with real values from the TTN control panel!"
```

rexfue, 2020-04-13:
Don't know how to get rid of this.

ThomasWaldmann, 2020-05-02:
It does not look like `FILLMEIN` is used anywhere in that file.

So just get rif of the whole if/else block there?

This needs to be tried on lora hardware, so I can't test this.

rexfue, 2020-05-02:
I can test that

rexfue, 2020-05-04:
Tested. Removed the whole if/else block end it works.
Also removed the unused variable sendjob and it works too.

ThomasWaldmann, 2020-05-04:
@rexfue sounds like a PR? :-)

commit 475c487e (2020-05-05): Merge pull request #252 from rexfue/remove_unneceessary_parts_in_lorawan.cpp

loraWan: removed  unused/not needed code

fixes #212 
fixes #234
closed 2020-05-05 by ThomasWaldmann