ecocurious2/MultiGeiger #235: code does not compile for wifi kit 32 any more

ThomasWaldmann, 2020-04-13:
i wanted to flash via usb and set the arduino ide to `heltec wifi kit 32`.

the code does not compile any more due to missing LoRa pin definitions (e.g. `DIO2`).

wanted to go the usb way because i needed usb logging anyway to debug some code.

besides that, one needs to compile for correct hw platform once to get correct partition layout.

ThomasWaldmann, 2020-04-13:
I suggest we:
- create a branch hardware_decision from current master
- reset master to cdf88e193158e77a366d99b4b8b5de186c662290 (last changeset before the hardware_decision work)

rexfue, 2020-04-14:
It compels, if you set CPU to Heltec wireless stick in Arduino ide, as is written in the read.me. The nit compiles for both hardwares . See readme PR ! It also does the right partitioning.

ThomasWaldmann, 2020-04-14:
Just setting the IDE to wireless stick also sets it to 8MB flash size and no OTA. That for sure won't work.

For everything else I guess we do not have documentation...

It **might** maybe work when also setting to 4MB flash size and "Minimal SPIFFS", but I didn't try that yet.

rexfue, 2020-04-14:
That **does** work. Ist sets partition to default 8MB, the means 4MB app and 4MB OTA.See compilation result: 
Sketch uses 1034898 bytes (30%) of program storage space. Maximum is 3342336 bytes.
And also the default_8MBV.csv:
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x330000,
app1,     app,  ota_1,   0x340000,0x330000,
spiffs,   data, spiffs,  0x670000,0x190000,
There are to app partitions of same size, the second used for OTA.

ThomasWaldmann, 2020-04-14:
A 8MB partition layout (intended for wifi stick device) obviously does not fit on a 4MB flash (like present on a wifi kit device).

rexfue, 2020-04-14:
Doch, der WiFi Kit hast auch 8 MB. Teste das mit dem esptool:
esptool.py --chip esp32 flash_id
Da kommt dann: Detected flash size: 8MB

ThomasWaldmann, 2020-04-14:
```
$ esptool.py --chip esp32 flash_id
esptool.py v2.5.1
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: 3c:71:bf:4d:a5:e4
Uploading stub...
Running stub...
Stub running...
Manufacturer: ef
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

rexfue, 2020-04-14:
OK, then buy a new one. The new ones have 8 MB!!

ThomasWaldmann, 2020-04-14:
That doesn't solve the issue for everybody who has a 4MB one.

rexfue, 2020-04-14:
I'd risk it - there are only a few. And they can live with the current version.
On the other hand you can of course try to solve the problem - I don't have time for that right now.

ThomasWaldmann, 2020-04-14:
Solution (Arduino IDE):
- Board: **Heltec Wifi Stick**
- Flash size: **4MB (32Mb)**
- Partition Scheme: **minimal SPIFFS (large APPS with OTA)**

```
$ cat ~/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/partitions/min_spiffs.csv 

# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
spiffs,   data, spiffs,  0x3D0000,0x30000,
```

As seen at the offsets/sizes used, the "minimal SPIFFS (large apps with OTA)" is a layout for **4MB** flash devices.

rexfue, 2020-04-14:
OK, that's a possible solution. But the we have two different binaries again :-(

ThomasWaldmann, 2020-04-14:
Does the .bin file exported by the Arduino IDE include a FULL image?

My impression was that it rather only has the contents for the code partition?

And that an OTA update using that .bin file **only** updates the code in 1 OTA partition.

rexfue, 2020-04-14:
No, don't think so. Arduino .bin is only the app. Therefore the long long line that the ide uses to flash (boot partition, partition scheme, app ...).

PR #238 (2020-04-14): docs: update development/release docs
how to create and test binaries, IDE settings, ...

Fixes #235.
closed 2020-04-14 by ThomasWaldmann