ecocurious2/MultiGeiger #240: feature flags

ThomasWaldmann, 2020-04-14:
It just came to my mind how we could implement feature flags with much less UI and much less work to implement that UI.

Currently we have these feature flags:
- speaker tick
- led tick
- display on
- startup sound

Considering #106 we might also need a "rgb led" feature flag.

To avoid having to add 1 web form element per feature (which might be more work than the feature itself), we could just have 1 string form element, like:
```
Features: display,tick,melody,led,...
```

In the software, we would just split that string and checking for a feature flag is then just:
```
display_enabled = 'display' in feature_list
```

ThomasWaldmann, 2020-04-14:
If we want something more powerful, we could even use `argparse` and have that feature string look like command line arguments / options.

rexfue, 2020-04-14:
Sounds nice. But how should the string be entered correctly on the small smartphone keyboard? And isn't that a bit anachronistic? Today everything should be clickable, right?

ThomasWaldmann, 2020-04-14:
I also don't like typing on a smartphone, but there are people who typo tons of messages on smartphones...

jensliebau, 2020-04-16:
Is a feature list sufficient? 

Of cause rebuilding WebUI for each and every feature is a lot of work. But on the pro side information could be structured. 

Enable Dislay, Tick or even RGB LED (#106) might be just ja checkbox, there might be more complex UI necessary for other features (#214) like "An eigene API senden" or "Senden an InfluxDB" would require more than one input field wich need to be grouped.