ecocurious2/MultiGeiger #248: check GM pulse software debouncing

ThomasWaldmann, 2020-05-03:
from the GM ISR code (shortened / simplified):
```
  timestamp_us_prev = timestamp_us;
  timestamp_us = micros();
  if ((timestamp_us - timestamp_us_prev) > GMC_DEAD_TIME) {
    // the rest is only executed if GMC_DEAD_TIME is exceeded.
    // Reason: pulses occurring short after another pulse are false pulses generated by the rising edge on the PIN_GMC_COUNT_INPUT.
    // This happens because we don't have an Schmitt-Trigger on this controller pin.
    ...    // count the pulse
    timestamp_us_prev_used = timestamp_us;
  }
```

I think this is wrong, because it is ignoring all pulses after the "previous" pulse for DEAD_TIME, no matter whether the "previous" pulse was considered valid ("prev_used") or not ("prev").

Correct would be to ignore all pulses after "previous used".

If there is only 1 false pulse after a valid one, it does not make a difference, but if there were multiple false pulses after a valid one, we would ignore pulses for longer than DEAD_TIME after the real one due to this bug.

ThomasWaldmann, 2020-05-03:
@rexfue @WwwBoehriDe ^^^

ThomasWaldmann, 2020-05-03:
#247 should be reviewed / merged before fixing this.

WwwBoehriDe, 2020-05-03:
OK, I have looked into this. I guess the problem will be fixed if the Line:
  isr_count_timestamp_us_prev = isr_count_timestamp_us;
is replaced by
  isr_count_timestamp_us_prev = isr_count_timestamp_us_prev_used;

commit 38937c71 (2020-05-07): Merge pull request #250 from ThomasWaldmann/fix-gm-debouncing

refactor pulse counting, fixed debouncing, fixes #248
commit 801c3f9d (2020-05-07): refactor pulse counting, fixed debouncing, fixes #248

fix: dead time computation was wrong: the dead time starts after last
VALID pulse, but the code was comparing to ANY last pulse.

refactor: used simpler variable names, got rid of 1 unneeded variable,
made deadtime condition easier to read.
closed 2020-05-07 by ThomasWaldmann