ecocurious2/MultiGeiger #257: display status

ThomasWaldmann, 2020-05-09:
maybe the OLED display could briefly show:
- wifi status (off, connecting, connected)
- ble status (off, ...)
- high voltage status (ok, error)
- transmission status (ok, error)

anything else?

ThomasWaldmann, 2020-05-10:
Maybe we could have something like:
```
int status[8];
```

On the OLED display that would map to a one line display like:
```
<S0> <S1> <S2> <S3> <S4> <S5> <S6> <S7>
```

If <S0> represents the WiFI status if could display:
`W-` for off (status[0] == 0)
`W?` for connecting (== 1)
`W+` for connected (== 2)
`W!` for error (== 3)
`  ` for do not display (== -1) - use this for reserved, not yet used status indicators.

Idea: always use same space for an indicator, so nothing shifts around when status line is updated.

That way, we could implement that once, without having to test again and again whether it works for all supported displays.

rexfue, 2020-05-10:
Very good idea. 
But on the little LoRa display we have only 8 characters per line, so there are only 4 statuses possible. Maybe this is enough for now.

ThomasWaldmann, 2020-05-10:
8 works for both, see PR #258.

PR #258 (2020-05-11): implement status display
fixes #257. 

slightly improves on #201 and #202.
commit 12a753eb (2020-05-11): implement status display, fixes #257
closed 2020-05-11 by ThomasWaldmann