ecocurious2/MultiGeiger #268: tries to send after reset w/o connected to WiFi

rexfue, 2020-05-11:
after reset, the program starts more or less immediately sending to the servers, not waiting for wifi connection. This happens not at every reset, but on many (like, every other time).
I'll do a PR to wait for wifi before sending

ThomasWaldmann, 2020-05-11:
Maybe just rearrange the `wifi_status` code inside the `loop`.

Only call `transmit` if `wifi_status` or TBD `lora_status` indicates a connection.

ThomasWaldmann, 2020-05-11:
guess this might be a timing effect. before the HV charging changes, the initial charging ran within main code (and took some time to complete - maybe enough so the wifi usually had connected until afterwards).

now the HV charging always runs in background, so it reaches the transmit code earlier.

rexfue, 2020-05-11:
Yeah, that may be the reason. PR is out

ThomasWaldmann, 2020-05-11:
the `transmit` code in multigeiger.ino waits one `MEASUREMENT_INTERVAL` (150s) before sending anything. so, this issue should not happen at all except if you wifi takes longer than that to connect.

rexfue, 2020-05-11:
No, it doesn't look like that.  Measured about 2.5 sec:
```
Setting up AP: ESP32-16752040
Use password: ESP32Geiger
AP IP address: 192.168.4.1
State changed from: 0 to 2
GEIGER: -------------------------------------------------------------------------------------------------
GEIGER: Simple Multi-Geiger, Version V1.13.1-dev
GEIGER: -------------------------------------------------------------------------------------------------
GEIGER: GMC_counts Time_difference Count_Rate Dose_Rate HV Pulses Accu_GMC Accu_Time Accu_Rate Accu_Dose
GEIGER:   [Counts]            [ms]      [cps]   [uSv/h]       [-] [Counts]      [ms]     [cps]   [uSv/h]
GEIGER: -------------------------------------------------------------------------------------------------
GEIGER: current time: 1970-01-01T00:00:01
GEIGER: Measured GM: cpm= 0 HV=0
GEIGER: Sending to Madavi ...
GEIGER: http request body: 
{
 "software_version": "V1.13.1-dev",
 "sensordatavalues": [
  {"value_type": "Si22G_counts_per_minute", "value": "0"},
  {"value_type": "Si22G_hv_pulses", "value": "0"},
```

ThomasWaldmann, 2020-05-11:
I don't see 2.5s there.

rexfue, 2020-05-11:
stopped time from reset to sending

ThomasWaldmann, 2020-05-11:
But maybe the `static int last = millis()` does not work like I assumed it does.
Because of the `static` I assumed that millis() call is done **once only** and **at program start time**, not at function call time.

rexfue, 2020-05-12:
And what do we do now? Can you fix that? So I can delete my PR then.

ThomasWaldmann, 2020-05-12:
I read some language docs and they confirmed that local static vars (should) work as I assumed.
I'll try to find out more later today.

rexfue, 2020-05-12:
👍

PR #297 (2020-05-15): check wifistatus in transmissions
replaces #270 fixes #268
closed 2020-05-15 by ThomasWaldmann