ecocurious2/MultiGeiger #289: optionally switch off WiFi

ThomasWaldmann, 2020-05-13:
When using the multigeiger with LoRa and also (in future) in mobile application, we sometimes do not need active WiFi or even want to switch it off to save some power consumption.

https://github.com/espressif/arduino-esp32/issues/1077

https://github.com/prampec/IotWebConf/issues/5

(from #180)

rexfue, 2020-05-14:
Yes, also think we schould do this. 
But I don't see how to do it with iotwebconf, the given link in your comment isn't really enlightening for me.

t-pi, 2020-05-14:
Maybe we can change this within release 1.15 and BLE integration, to allow (basic...) configuration alternatively via BLE before switching of Wifi. I guess one interface needs to stay open...

ThomasWaldmann, 2020-05-14:
@rexfue it sounded like the iotwebconf author waits for a contributor PR to actually implement this.

ThomasWaldmann, 2020-05-14:
@t-pi adding config via BLE might be quite some work, right? and might work very differently because that is not support by iotwebconf.

so I'ld suggest always having WiFi initially on even if the wifi setting says it should be off - just to give the user the chance to connect to the multigeiger esp32 accesspoint (for once, for a minute) and change settings.

rexfue, 2020-05-14:
Yes, we always should allow to connect to the internal AP after reset for a certain  time.