ecocurious2/MultiGeiger #295: Status display after reset

rexfue, 2020-05-14:
Should the characters for the servers (s, m, t) not appear directly after reset if enabled?

ThomasWaldmann, 2020-05-14:
No, the initial state is "NODISPLAY".

After first transmission attempt (150s), the status will update to either IDLE (s, m, t) or ERROR (some number).

See also the current docs on RTD.

I first had it differently, but only with that new method the error number is visible for a longer time (which is important so people actually notice it).

ThomasWaldmann, 2020-05-14:
Guess I need to add "not used yet" to the description of `.` state.

rexfue, 2020-05-14:
OK, I see. But I think it would be better to see right away if the configuration is correct, e.g. if 'send to madavi' is enabled. Maybe you could do this with another character: dash (-) instead of the dot as placeholder and the dot (.) , if enabled by configuration.
Error handling btw. is OK.

ThomasWaldmann, 2020-05-14:
When only displaying a `-` it is a bit hard to see what that means (you only have the position as indication then and need to either remember or look it up in the docs or wait until it shows the idle/error status).

But maybe it could be set to idle state once via `setup()` (if enabled).

It just must not be periodically overwritten in `loop()`.

ThomasWaldmann, 2020-05-14:
Hmm, no, we can't do it in `setup()` because we do not have a WiFi connection there and do not know whether we can establish one later. I would not like to show idle `s` or `m` as long as I can't be sure WiFi is working.

ThomasWaldmann, 2020-05-14:
I'll just display "?" if enabled, but as long as status is unknown...

commit 25601b46 (2020-05-15): Merge pull request #302 from ThomasWaldmann/status-init-state

status display: add init status ("?"), docs, fixes #295
commit c3bcba25 (2020-05-15): status display: add init status ("?"), docs, fixes #295
closed 2020-05-15 by ThomasWaldmann