ecocurious2/MultiGeiger #307: monitor / display wifi and lora signal strength

ThomasWaldmann, 2020-05-16:
this might be useful for positioning / directing the multigeiger.

problem: OLED is not visible when putting the whole thing into the grey plastic tube.

so only these are left to display status:
- wifi and builtin web interface
- usb log output

ThomasWaldmann, 2020-05-16:
Hmm, is this technically possible?
- We can only access the web status page when the AP is active.
- But we can only know WiFi signal strength while the wifi client is active.

rexfue, 2020-05-16:
No, we can see the web page at every time: point your browser to the ip address of the geiger ( or to http:// ESP32-xxxx with xxx the chip id)

ThomasWaldmann, 2020-05-16:
Ah, right, but that's going over the wifi client connection then and you can't rely on this if you have:
- connection troubles to uplink wifi AP
- no wifi uplink, only LoRa

t-pi, 2020-05-16:
Hmm, a fully blown config interface via BLE is indeed some time away, but some additional info might be relatively easy to implement. Which info would be required, and is it already available?

ThomasWaldmann, 2020-05-16:
Maybe keep the BLE stuff to existing functionality for now. This ticket is about not yet existing stuff. :-)

See also #308.

M-a-r-i-a-n-n-e, 2021-03-21:
@ThomasWaldmann I would prefer the same solution in grafana as for all sensor.community devices: https://api-rrd.madavi.de/grafana/d/Fk6mw1WGz/wifi-signal?orgId=1&var-chipID=esp8266-4206630 (I just selected an arbitrary ESP).