ecocurious2/MultiGeiger #316: Initial configuration missing -> no Geiger counter functionality

t-pi, 2020-05-21:
I just did an update from 1.12 to 1.15-dev on a workshop device that has not been set up, and it was stuck in the welcome screen. That is not a great user experience, especially at the workshops, where there might be no Wifi...

I guess this comes from the new NTP functionality:
```
  // a bug in arduino-esp32 1.0.4 crashes the esp32 if the wifi is not
  // configured yet and one tries to configure for NTP:
  while (iotWebConf.getState() <= IOTWEBCONF_STATE_NOT_CONFIGURED) {
    iotWebConf.delay(200);
  }
  setup_clock(0);  // 0 == do NTP!
```
Could we somehow async the NTP setup?

t-pi, 2020-05-21:
...related: The README still refers to the "connecting..." status line.

ThomasWaldmann, 2020-05-21:
We can move that to main loop.

As the comment right below these lines says, the clock sync happens later and async anyway.

commit bd041eb7 (2020-05-22): do async NTP/clock setup, fixes #316
commit d29d2f17 (2020-05-22): Merge pull request #333 from ThomasWaldmann/non-blocking-ntp-setup

do async NTP/clock setup, fixes #316
closed 2020-05-22 by ThomasWaldmann