ecocurious2/MultiGeiger #323: Deutsche Uebersetzung

ThomasWaldmann, 2020-05-22:
https://multigeiger.readthedocs.io/de/latest/ vs https://multigeiger.readthedocs.io/en/latest/

Man sieht: da fehlt noch viel, sowohl englisches Original als auch deutsche Uebersetzung.

Uebersetzen kann man dort: https://www.transifex.com/thomaswaldmann/multigeiger

Datenfluss:
en *.rst --> *.pot --> push to transifex
(translate)
pull from transifex --> de *.po --> git / github --> readthedocs

ThomasWaldmann, 2020-05-22:
Am wichtigsten sind natuerlich der Index (README-en.rst) und alles, was einfache Anwender brauchen.

Zeug fuer SW-Entwickler und sonstige Nerds ist nachrangig, die verstehen meist auch Englisch.

ThomasWaldmann, 2020-05-22:
README-en ist fertig.

Marwe, 2020-12-28:
I translated all Text in transifex to DE, left out `docs_build_gettext_changes`, because that seems to be difficult and unneccessary to translate anyway.

Pull needed from transifex, as documented in
https://github.com/ecocurious2/MultiGeiger/issues

Marwe, 2020-12-28:
Some notes:

* Dokumentation
    * PDF, ODT, ODS to rst
    * add internal section
    * README to EN, reduce to pitch only

ThomasWaldmann, 2020-12-29:
@Marwe thanks for translating. I reviewed most of the strings, did some minor fixes here and there and updated master branch on github with the translations.

Baitcookie, 2020-12-29:
@ThomasWaldmann. Sent you a join request at transiflex

ThomasWaldmann, 2021-03-27:
superseded by the more general #390.

closed 2021-03-27 by ThomasWaldmann