ecocurious2/MultiGeiger #334: BLE compiler warning - can we get rid of it?

ThomasWaldmann, 2020-05-22:
```
/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/BLE/src/BLERemoteService.cpp: In member function 'void BLERemoteService::getCharacteristics(std::map<short unsigned int, BLERemoteCharacteristic*>*)':
/home/user/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/BLE/src/BLERemoteService.cpp:246:89: warning: parameter 'pCharacteristicMap' set but not used [-Wunused-but-set-parameter]
 void BLERemoteService::getCharacteristics(std::map<uint16_t, BLERemoteCharacteristic*>* pCharacteristicMap) {
```

t-pi, 2020-05-22:
Hmm, can't reproduce. 
Do you have set different build flags in your platform.io?

ThomasWaldmann, 2020-05-22:
I don't use PlatformIO, but Arduino IDE "Compiler Warnings: All".

t-pi, 2020-05-22:
Hmm, Version # of ESP32 BLE lib in platform.io seems to be 1.0.1 / Espressif 32 platform 1.12.1, both do not match your 1.0.4 version. Here in platform.io, the relevant code seems to be this one:
```
/**
 * @brief This function is designed to get characteristics map when we have multiple characteristics with the same UUID
 */
void BLERemoteService::getCharacteristics(std::map<uint16_t, BLERemoteCharacteristic*>* pCharacteristicMap) {
	pCharacteristicMap = &m_characteristicMapByHandle;
}  // Get the characteristics map.
```
Any idea, what could be wrong there?

ThomasWaldmann, 2020-05-23:
Not sure if I am reading it correctly, but the code inside that function does not make sense to me.

It assigns a new value to a pointer that was just local (because it is a function parameter).

t-pi, 2020-05-23:
Just discovered an alternative, more light-weight BLE lib, forked from the original:
https://github.com/h2zero/NimBLE-Arduino
Will check their claims of an easy transition from ESP32 BLE lib...

t-pi, 2020-05-25:
First update: Transfer to NimBLE is quite straightforward, and the (positive) effect on memory is indeed impressive:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.5% (used 60780 bytes from 327680 bytes)
Flash: [====      ]  39.1% (used 1305666 bytes from 3342336 bytes)
```
However, the lib is at version 0.9.0, heading for 1.0 and has even more compiler warning (like, lots...)
```
warning: pointer of type 'void *' used in arithmetic [-Wpointer-arith]
```
So for the moment I'll keep it in a local branch, see if that maybe impacts the runtime (which is currently not great, as discussed elsewhere) and wait for v1.0.x...

ThomasWaldmann, 2021-03-25:
can't reproduce any more...

closed 2021-03-25 by ThomasWaldmann