ecocurious2/MultiGeiger #336: BLE: need simple FOSS app

ThomasWaldmann, 2020-05-22:
I tried 3 different android FOSS apps for *runners*, but didn't get any display of "heart rate" (cpm) working.

Only reaction that something might work is the `b` changing to `B`.

ThomasWaldmann, 2020-05-22:
see also #330.

ThomasWaldmann, 2020-05-22:
FOSS == on f-droid or apk download from trusted site.

Can't use google play.

t-pi, 2020-05-22:
Plan is to have our own simple app, soon-ish.
But while testing, please keep in mind that we send an update packet only every ~10s, with even more pause if Wifi transmission is blocking once again. So it might take a while until you see a reaction in the app with actual data.
Good point, will add that to the docs :)

t-pi, 2020-07-02:
Hi @ThomasWaldmann, my repo with a Flutter app for the multigeiger is on github, however right now still as private repo. I have invited you in, too. It shall become public asap, however as it is my first "larger" project, I'd like to discuss with some guys like you whether it's missing some important stuff, has maybe critical issues, etc.

ThomasWaldmann, 2020-07-03:
OK, I'll have a look!

t-pi, 2021-03-25:
Here's the first public version:
https://github.com/t-pi/multigeiger_companion/tree/0.1

ThomasWaldmann, 2021-04-09:
TODO/CHECK: do we have a link to that repo from our bluetooth docs?

MarcelWaldvogel, 2021-04-24:
[`Hrm`](https://git.wow.st/gmp/hrm) works for me. Just installed it from F-Droid.

I noticed that sometimes it can't reconnect, but that may also be a problem on the ESP side (haven't looked into it closer yet).

MarcelWaldvogel, 2021-04-29:
The reconnection seems to be an app issue, not a Multigeiger issue. Disconnect/Reconnect in the app seems to be necessary sometimes after walking out of the reception range (not always, though).

I therefore think no ESP BLE changes are needed at this time.