ecocurious2/MultiGeiger #346: Low power future: Use ULP for interrupt counting?

t-pi, 2020-05-23:
If we discuss switching off WiFi to reduce low power, would it (at a later stage) be possible to use the ULP coprocessor to count the GM impulses and only read them every <display update interval> seconds? I think we discussed this already once in an offline meeting.
@ThomasWaldmann, [you seem to have experience](https://github.com/ThomasWaldmann/py-esp32-ulp) already...

ThomasWaldmann, 2020-05-23:
I don't really have much experience with the ULP besides writing this assembler in python.

Not sure whether we can use the ULP for GM counting - it might be a bit too simple / too slow / too limited for that.

ThomasWaldmann, 2020-05-23:
An alternative approach to power saving could be to let the device sleep most of the time and only wake up for measurements and transmission of results.

If one assumes that radiation in nature does not change rapidly usually, this might be good enough for normal times, but has issues in crisis times (like losing time in sleep mode until a higher radiation level is measured).