ecocurious2/MultiGeiger #348: use hw pulse counter for GM events?

ThomasWaldmann, 2020-05-23:
The ESP32 has hardware pulse counters (PCNT), which we maybe could use:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/pcnt.html

Noticeable:
- 16bit
- count on rising or falling edge
- the inputs have optional filters so that pulses shorter than X can be ignored.

With low pulse rates (as with natural radiation levels), I guess we would not really gain something.

Not sure about how well our code copes with high pulse rates, see #280.

ThomasWaldmann, 2020-05-23:
Hmm, we can't trigger the `tick` sound per GM pulse if we count in hardware and do not have a sw interrupt per pulse.

t-pi, 2020-05-24:
This could be another approach for #346: Let the HW counter count until we wake up, display, notify, transmit and go back to sleep. Indeed, this wouldn't work with speaker / LED ticks, but for outdoor use that's maybe less needed and can be made conditional. 
In my case, speaker ticks are always off, LED stays on, but only because it doesn't disturb, not because I need it...

ThomasWaldmann, 2020-05-25:
Not sure whether the PCNT is powered in deep sleep:
```
In deep sleep mode, CPUs, most of the RAM, and all the digital
peripherals which are clocked from APB_CLK are powered off.
The only parts of the chip which can still be powered on are:
RTC controller, RTC peripherals (including ULP coprocessor),
and RTC memories (slow and fast).
```