ecocurious2/MultiGeiger #351: BLE: Properly initialize stored config

t-pi, 2020-05-25:
During NimBLE testing I noticed this in the startup-message:
```
Loaded config 'send2Community'= '1'
Loaded config 'send2Madavi'= '1'
Loaded config 'send2ble'= '��0'
```
My guess is, that the local storage was initialized without the BLE parameter, as I was indeed not yet in the web config on that device since BLE update. How can this properly set up to avoid random memory output?

rexfue, 2020-05-26:
The only way I know is to change the CONFIG_VERSION. But then ALL variables are deleted including the wif icredentials. This means that with OTA update the user has to go to the device and dial into the Geiger API after restart to enter his credentials again.
Or we write in the readme that the new configuration variables are uninitialised and have to be entered. 
Both is not ideal, but I don't know what else.

ThomasWaldmann, 2020-05-28:
There could also be a validation step (when reading config from nvram, when getting form data from user), that ensures valid values (and in invalid case, uses default value or asks again).

t-pi, 2020-05-28:
Asking is difficult, as we might not have a display on, but from userdefines.h we have valid initial values which just need to be stored.

In the code it looks like first initConfigVariables initalizes from userdefines, but the status is then overwritten with loadConfigVariables from the EEPROM. Could we check for discrepancies here and initialize with userdefines and a iotWebConf.configSave() in case of difference?
In the webinterface, it indicates correctly "0/1" in case of undefined, but as we point to a boolean and read a string(?), this might be a crash or other security risk...

rexfue, 2020-05-28:
How do we recognize that the data in the EEprom is garbage and not just stored user data? If we only compare to userdefines.h there could/would be a difference at every boot process.

ThomasWaldmann, 2020-05-28:
If the valid values are e.g. just 0 and 1 and we see some different value, we know it is invalid.

Of course we can not know whether the user has set a valid value or it was just valid by chance.

t-pi, 2020-05-28:
The default setup in my case was '��0' and in the config interface it was shown as greyed '0/1', so probably some represenataion of null. Only these cases need to be handled from userdefines, so the writes to config should be very limited.
And in all these cases the user has not yet decided, except maybe via the switches, but they override the config anyways.

ThomasWaldmann, 2020-05-28:
"asking" in my comment about validation referred to the case when the user POSTs the form. if a valid is detected as invalid, we can just show the form again and ask for valid input.

t-pi, 2020-05-28:
Ah okay! My aim was to get rid of these uninitialized states right from the (program-)start, to avoid the ugly non-printable character output. But indeed, I just tested it, in the UI you can just delete the value and store the 'null'. So yes, we'd need some input validation, too...