ecocurious2/MultiGeiger #352: Info to users about recurring reboots

t-pi, 2020-08-05:
In the Slack, there's currently a discussion about (among other topics) the reboots with a (non-developer) user. It would be good to mention the current recurring reboots in our documentation about V1.14, wouldn't it?

M-a-r-i-a-n-n-e, 2021-03-20:
@ThomasWaldmann  @WwwBoehriDe Is there already a workaround or a fix for this issue?

ThomasWaldmann, 2021-03-20:
In the last BBB meeting, we noticed that there is arduino-esp32 1.05 out now since ~3 weeks.

Maybe they fixed the problem in the updated HTTPClient code.

M-a-r-i-a-n-n-e, 2021-03-20:
I don't think so, as I already compiled it with framework-arduinoespressif32 3.10005.210308 (1.0.5) cf. https://github.com/espressif/arduino-esp32/releases

M-a-r-i-a-n-n-e, 2021-03-20:
Is there any workaround? Is the root cause already known?

ThomasWaldmann, 2021-03-20:
See #314. This issue is just about writing some docs, notifying the users - not about the crash and its root cause.

ThomasWaldmann, 2021-03-20:
I thought about this. If a user encounters an issue, expectation is that the issue tracker on github is checked whether the issue is known (which is the case here, see #314) and if not, file a new issue.

Thus, there is nothing to do here.

closed 2021-03-20 by ThomasWaldmann

M-a-r-i-a-n-n-e, 2021-03-20:
@ThomasWaldmann You think that a defect that leads to a restart every second is unworthy of a line in the documentation? Interesting.

ThomasWaldmann, 2021-03-21:
I don't have restarts every second.

ThomasWaldmann, 2021-03-21:
@M-a-r-i-a-n-n-e maybe you(?) have a different issue - then we need a new github issue with tracebacks similar to the ones you see in #314 (IF they are different from the ones in #314).

M-a-r-i-a-n-n-e, 2021-03-21:
> @M-a-r-i-a-n-n-e maybe you(?) have a different issue - then we need a new github issue with tracebacks similar to the ones you see in #314 (IF they are different from the ones in #314).

Yes, unfortunately this seems to be the case, although I just followed the simple steps in https://multigeiger.readthedocs.io/en/latest/development.html#flashing-devices-creating-binaries (and of course installed the libraries using the library manager).

ThomasWaldmann, 2021-03-21:
@M-a-r-i-a-n-n-e try that: https://github.com/ecocurious2/MultiGeiger/releases/tag/V1.15.0

M-a-r-i-a-n-n-e, 2021-03-21:
@ThomasWaldmann Thank you for the new release! It first hurt my ears, because as long as I didn't erase the ESP32 memory it beeps at full volume, but after erasing the memory with the ESP32 Download Tool, it runs like a charm! 🙂👍