ecocurious2/MultiGeiger #357: master branch broken due to IotWebConf

ThomasWaldmann, 2020-12-29:
https://travis-ci.org/github/ecocurious2/MultiGeiger/jobs/751911779

guess it's their 3.0 release breaking things...

ThomasWaldmann, 2020-12-29:
https://travis-ci.org/github/ecocurious2/MultiGeiger/jobs/751911779#L332 -> 3.0.0

https://github.com/prampec/IotWebConf/commit/4f97541e20155679c42fcf20222a3d7c75947882

ThomasWaldmann, 2020-12-29:
https://github.com/prampec/IotWebConf/blob/master/doc/MigrationGuide-v3.0.0.md

we need to follow this for our code, then we can remove the version restriction from `platformio(-example).ini` again.

rexfue, 2020-12-30:
I can work on the migration to 3.0 - OK?

ThomasWaldmann, 2020-12-30:
sure :)

ThomasWaldmann, 2020-12-30:
btw, the author told that 3.0 is not officially released yet (which is a bit strange as it is obviously automatically distributed already).

but we can already have a PR, so we can merge it after 3.0 is "officially released".

rexfue, 2020-12-30:
OK, then I test it locally and wait until official release to make a PR

ThomasWaldmann, 2020-12-30:
Please make PRs always as early as possible to get feedback/review.

ThomasWaldmann, 2021-05-15:
iotWebConf was upgraded to 3.x right after the 1.15.1 release, so it will be in 1.16.0 release.

closed 2021-05-15 by ThomasWaldmann