ecocurious2/MultiGeiger #366: Missing BOM / Clarification for Capcitors neede (Rastermass value)

umweltsensor, 2021-01-09:
Hi There

Wonderful project you have here, thanks for all the effort you all put in!

The original (and meanwhile outdated) BOM found here http://www.boehri.de/elektronik/MultiGeiger/MultiGeiger.xls used a ceramic cap. The list found in the Aufbauanleitung https://github.com/ecocurious2/MultiGeiger/blob/master/docs/Bauanleitung/Aufbauanleitung.pdf gave me some hints, but no specific model numbers are given.

So I am searching for an up to date BOM (Bill of Material). Is there a list with detailed specifications of all parts? I'm missing the specs of the condensators C1 to C4 especially for C2 (that is RM5 = 5mm pin spacing, measured in the eagle board file). 

For C2 would go for this one: https://www.reichelt.com/ch/de/folienkondensator-10nf-1000v-rm7-5-mks4-1000-10n-p172943.html?search=+10nf+1000v&&r=1

It a 10nF / 1000V WIMA capacitor with 7.5mm pin spacing probably fitting in the holes. Couldn't find a RM5 version of C2 on reichelt.

Cheers!

ThomasWaldmann, 2021-01-10:
such stuff should live in the git repo, preferably in a non-binary format.

umweltsensor, 2021-01-10:
OK, I'm on it. I'll create a txt tab delimited BOM and create a pull request.

umweltsensor, 2021-01-10:
Pull request created (https://github.com/ecocurious2/MultiGeiger/pull/369). Please check if nothing is missing or wrong.