ecocurious2/MultiGeiger #367: Add Multigeiger to kitspace.org

umweltsensor, 2021-01-09:
Would that be an idea to add the multigeiger on kitspace.org? If it's ok for you, i would start it and add your gerbers (directly via your github) to kitspace.org and create a bill of materials.

ThomasWaldmann, 2021-01-10:
note: some developers talked shortly about this at rc3 online conference in the context of #363 (kitspace is also used there).

ecocurious, 2021-01-14:
@umweltsensor Wir überlegen noch ob das Sinn macht, denn uns wäre es wichtig, dass das Gerät gut mißt (funktionierendes Messnetz). Das Meßnetz wird genauer, wenn wir ein paar standartisierte Bauteile haben (Gehäuse mit bestimmter Dicke ist beispielsweise wichtig).  Bin neugierig, wie Du auf Kitspace gekommen bist, betreibst Du die Plattform? Was für Umweltsensoren machst Du? Schreib mal ein paar Sätze zu Dir/Deiner Motivation, bin neugierig.

umweltsensor, 2021-02-17:
@ecocurious Kitspace habe ich mehr zufällig entdeckt; bin nicht der Betreiber. Die Gedanken zur Standardisierung kann ich nachvollziehen. Das führt irgendwann zu einer Abwägungsfrage, wie weit man citizen science, laiengerechte (Sensoren-)Projekte und Messgenauigkeit unter einen Hut bringt. Persönlich (und in meinem aktuellen Überlegungsstand) tendiere ich zu breitem Publikum mit dem Nachteil, dass darunter die Messgenauigkeit etwas leidet. Das grössere Publikum sehe ich als Chance dafür, dass mehr Köpfe die Projekte vorantreiben und in den Details verbessern. Und wer die Musse hat, kann auf bestehender Open Source neue und grössere Würfe hinlegen.

Ich selbst komme beruflich nicht aus der Elektrotechnik oder ähnlichem, sondern aus einem völlig anderen Bereich. Stecke aber in den paar freien Minuten gerne meine Zeit in solche Projekte. Ich habe vor, eine Infoplattform aufzustellen in EN/DE für konkrete bestehende Sensor-Projekte. Zudem arbeite ich an einer Kombination von airrohr, DNMS, Multigeiger etc. mit dem Ziel, eine möglichst simple Anleitung mit möglichst breit verfügbaren Komponenten zu erstellen, damit auch Leute, die nicht ständig mit Lötkolben hantieren, Sensoren aufstellen und Spass daran bekommen.

Die Frage nach dem Gehäuse und den Messwerten finde ich spannend. Hättest Du da gerade Infos oder den einen oder anderen Link zur Hand?

ecocurious, 2021-02-17:
Das klingt sehr gut. Wenn Du magst, lade ich Dich zu unserer nächsten Telko kommende Woche ein. Oder wir beide telefonieren vorab, denn mir ist das auch sehr wichtig. Wir haben da gerade etwas Chaos in sachen Doku und Auffindbarkeit. Schreib mir einfach ne Mail, ich suche Dir die Links zum Gehäuse etc. raus. Danke und liebe Grüße Nicola 

Ps: Die Messwerte siehst du, wenn du auf unserer Karte  https://ecocurious.de/multigeiger-karte/ <https://ecocurious.de/multigeiger-karte/>. auf einen Sensor klickst. 
Oder wie war Deine Frage gemeint?

ThomasWaldmann, 2021-03-25:
@WwwBoehriDe ^^^ bitte beantworten, das wuerde uns vielleicht viel erleichtern.