ecocurious2/MultiGeiger #371: Needed libraries? Can't find U8x8lib.h

hagenbuch, 2021-02-03:
Hi all,

somehow I miss a list of dependencies. When trying to compile, I could resolve the dependency to IoTWebConf because it is offerend in the "library" section. Yes, I avoided 3.x.y and installed 2.3.2

However 

`#include <U8x8lib.h>`

is not that obvious. I think I have to download the zip from here

`(https://github.com/olikraus/U8g2_Arduino)`

And it worked.

Now it complains on this line

`LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);  // g-band`

it says "DR_DF12" was not declared in this scope. I have your Heltec borad you gave us in Freiburg, it does not use LoRa.. 

I've been browsing the docs but to no avail..

Cheers and thanks
Andreas

rexfue, 2021-02-03:
@hagenbuch You can find the information in https://github.com/ecocurious2/MultiGeiger/blob/master/README-de.md. Unfortunately, we have not translated it into English yet.

M-a-r-i-a-n-n-e, 2021-03-14:
It is translated into English now: https://multigeiger.readthedocs.io/en/latest/installation.html
I've got the same error message first, but after installing the libaries in VScode using the Arduino Libary Manager, everything worked.
I think this issue could be closed.

closed 2021-03-14 by ThomasWaldmann