ecocurious2/MultiGeiger #372: powering device?

Morg42, 2021-02-05:
Is there a expressly preferred way to power the multigeiger?

The ESP kit has a two-wire cable (red/black) which seems to mount to the surface bracket, and it has a USB connector. On the main PCB are no markings for powering.

It seems to work when powered via the micro-usb connector. Is this the preferred power source?

This could be specified in the documentation, I think it is missing.

Not submitting PR as I don't know if this is the right way.

ThomasWaldmann, 2021-02-05:
Isn't there something on the parts list?

I personally just use a USB power supply connected via the micro-USB connector.

If the multigeiger is connected to the development machine via USB data cable, it also gets power via that cable.

Morg42, 2021-02-06:
I didn't find anything in the parts list. 

As I said, at the moment I'm powering it via USB which works. It should just be mentioned in the docs, especially as the ESP module has a spare cable which was misleading at first...

ThomasWaldmann, 2021-02-06:
I thought #122 had fixed that, can you verify?

Morg42, 2021-02-07:
The parts list includes some parts to supply the measuring station with 5V 100mA. Neither the assembly docs for the PCB nor the assembly docs for the measuring station mention explicitly how to power this up.

When using the ready-packaged set, you possibly don't even read the parts list.

If you're content with this - fine. As someone who had to try and find the docs in the first place, there is still some way to go. Just saying...