ecocurious2/MultiGeiger #373: LoRaWan Version 1.14.0 Failure

perpaul, 2021-02-15:
After installing above version on a Wireless Stick and after several loops the controller stopped with the error message "...\libraries\MCCI_LoRaWAN_LMIC_library\src\lmic\radio.c:1065". Research on google reveals that it may be a hardware problem. The Wireless Stick is new and out of the box. A second  stick shows the same error.
How can I fix the problem?
Best regards
Paul.

rexfue, 2021-02-17:
@perpaul To be able to help I need a little more information:  what do you mean with 'after several loops'? Where do you see this error message? And what is your IDE (arduino or platformio)?

perpaul, 2021-02-17:
We use Arduino IDE and see the errormessage on the serial console. By loop we mean the transmission to TTN after several measurements. The multigeiger continues to measure after the crash what we can be seen in the display and on the ticks. We are currently testing version 3.1 of the MCCI_LoRaWAN_LMIC_library to see whether this solves the problem.

rexfue, 2021-02-17:
OK.  We have never had this error. Maybe the new library helps. Please keep us up to date.

perpaul, 2021-02-21:
Following the response of Terry Moore, the developer of the MCCI LoRaWAN LMIC library for your info:

Looking at the sketch, I would say that you can remove the calls to LMIC_setupChanne(); they’re not needed, and will be overridden by TTN during join. Also, you can’t change channels 0-2 in EU868; the LMIC ignores this – it certainly won’t let you change frequencies or bands, and at least one of 0-2 is BAND_MILLI. But this is not your problem.
LMIC_setTxData2() returns errors (it returns lmic_tx_error_t); please check the docs, and use its results. The logic in lorawan_send() is trying to duplicate work that the LMIC already does, and therefore is risky. In older versions of the LMIC, it might have been required, but with v3.3.0 it’s not advisable or required. In my experience, the LMIC is now very good about always either transmitting your data or reporting an error status.
I would not call LMIC_setLinkCheckMode(0) – in most cases the device and the network works better if linkcheck mode is enabled. In addition, it’s very important for handling the TTN V2 -> V3 transition. Furthermore, the comment “slow data rates change max TX size” misses the key point about v3.3.0 – the LMIC automatically increases the DR on transmit if the message is too big for the current DR. (If this is not desirable, you must call LMIC_setTxData2_strict().)
Setting LMIC.dn2Dr is normally not required.
If you’re going to call LMIC_setDrTxpow(), it should be called after join complete; and it’s not a bad idea to do so.
However, the key point is this:
Always call os_runloop_once() in your loop() routine. This must be done without fail, because the MAC can keep going (and will keep going) even after a transmit completes. As far as I can tell, the code only does this while a TX is pending. Not a safe optimization because the network can keep the LMIC busy arbitrarily long. (A lot longer, I must add, in TTN V3, because the RX window is now 5 seconds – each mac downlink can cause a mac uplink, which means things can go on for quite a while.)

perpaul, 2021-02-21:
Here an additional post from Terry:

I see that the opmode is 0x118 at the point of failure; this is a rare case where the debug prints are helpful (though it’s better to print these outside the LMIC). This means that a poll was pending (bit 4 is set, OP_POLL).
I think this means that the state tracking (in onEvent()) is getting confused.
This can happen if you try to send while the LMIC is busy dealing with downlink messages.
This can in turn happen after a lot of uplinks, because TTN may decide to send MAC downlinks. But the LMIC sends EV_TXCOMPLETE as soon as it knows your message has been sent and the RX slots have been process. This may be before mac processing is complete. So EV_TXCOMPLETE can’t be used as evidence that the mac is ready for a new message.
The comment says that you’re not honoring TTN fair use, so I gather that you might be transmitting fairly often?
I think you may have found a bug in the LMIC; it appears to me at the moment that LMIC_setTxData() needs to check whether any of (OP_POLL | OP_TXDATA | OP_TXRXPEND) are set.
So you should try:
if (LMIC.opmode & (OP_POLL | OP_TXDATA | OP_TXRXPEND)) {
   // don't transmit, LMIC is probably busy

commit 2e689053 (2021-03-04): Merge pull request #377 from terrillmoore/lorawan-pr-tmm

Fix #373: Improve LoRaWAN stability
closed 2021-03-04 by rexfue