ecocurious2/MultiGeiger #379: Reflashed device - no connection to AP possible

dl2sba, 2021-03-05:
I've reflashed my Multigeier and I'm no longer able to connect to the AP as well the ESP32 didn't connect to my WLAN.

```
19:43:28.728 -> GEIGER: Logging initialized at level 0.
19:43:29.147 -> GEIGER: Switches: SW0 speaker_on 1,  SW1 display_on 1,  SW2 led_on: 1,  SW3 ble_on: 1
19:43:29.194 -> GEIGER: BME_Status: not found
19:43:29.194 -> Config size: 179
19:43:29.194 -> Wrong config version.
19:43:29.194 -> E (688) MCPWM: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/mcpwm.c:190 (mcpwm_set_duty_type):MCPWM DUTY TYPE ERROR
19:43:29.194 -> E (695) MCPWM: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/mcpwm.c:190 (mcpwm_set_duty_type):MCPWM DUTY TYPE ERROR
19:43:29.241 -> GEIGER: -------------------------------------------------------------------------------------------------
19:43:29.241 -> GEIGER: Simple Multi-Geiger, Version V1.15.0-dev
19:43:29.241 -> GEIGER: -------------------------------------------------------------------------------------------------
19:43:29.288 -> GEIGER: GMC_counts Time_difference Count_Rate Dose_Rate HV Pulses Accu_GMC Accu_Time Accu_Rate Accu_Dose
19:43:29.288 -> GEIGER:   [Counts]            [ms]      [cps]   [uSv/h]       [-] [Counts]      [ms]     [cps]   [uSv/h]
19:43:29.288 -> GEIGER: -------------------------------------------------------------------------------------------------
19:43:29.336 -> AP password was not set in configuration
19:43:29.336 -> State changing from: 0 to 1
19:43:29.429 -> Setting up AP: ESP32-7936376
19:43:29.429 -> With default password: <hidden>
19:43:29.429 -> AP IP address: 192.168.4.1
19:43:29.429 -> State changed from: 0 to 1
19:43:40.493 -> GEIGER:         17           11587   1.467161  0.119483        27       17     11587  1.467161  0.119483
19:43:40.586 -> Connection to AP.
19:43:44.604 -> Disconnected from AP.
19:43:50.475 -> GEIGER:         20           10160   1.968504  0.160312         3       37     21747  1.701384  0.138558
19:44:00.482 -> GEIGER:         18            9860   1.825558  0.148671         1       55     31607  1.740121  0.141713
19:44:10.456 -> GEIGER:         20            9884   2.023472  0.164789         2       75     41491  1.807621  0.147210

```
I've tried it with an iPhone and a WindowsPC to connect to the AP - no success :-(

Any ideas?

//Dietmar

rexfue, 2021-03-05:
@dl2sba:  Which version did the multigeiger have before? The 1.15. is a developer version. Which IDE did you use to compile this? When did you download the source code?
Maybe, you use the wrong password?  We can discuss further by PM (rexfue@gmail.com), I will summarize this here later.

ThomasWaldmann, 2021-03-06:
```
19:43:29.336 -> AP password was not set in configuration
19:43:29.429 -> Setting up AP: ESP32-7936376
19:43:29.429 -> With default password: <hidden>
```

Tried that ESSID with the default password?

dl2sba, 2021-03-06:
Now used 1.14.0 from releases 

```
04:53:34.645 -> rst:0x1 (POWERON_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
04:53:34.645 -> flash read err, 1000
04:53:34.645 -> ets_main.c 371 
04:53:35.015 -> ets Jun  8 2016 00:22:57
04:53:35.015 -> 
04:53:35.015 -> rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
04:53:35.015 -> configsip: 0, SPIWP:0xee
04:53:35.015 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
04:53:35.063 -> mode:DIO, clock div:1
04:53:35.063 -> load:0x3fff0018,len:4
04:53:35.063 -> load:0x3fff001c,len:1044
04:53:35.063 -> load:0x40078000,len:8896
04:53:35.063 -> load:0x40080400,len:5816
04:53:35.063 -> entry 0x400806ac
04:53:35.620 -> GEIGER: Logging initialized at level 1.
04:53:36.080 -> GEIGER: BME_Status: not found
04:53:36.080 -> Config size: 177
04:53:36.080 -> Wrong config version.
04:53:36.080 -> E (671) MCPWM: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/mcpwm.c:190 (mcpwm_set_duty_type):MCPWM DUTY TYPE ERROR
04:53:36.080 -> E (678) MCPWM: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/mcpwm.c:190 (mcpwm_set_duty_type):MCPWM DUTY TYPE ERROR
04:53:36.125 -> AP password was not set in configuration
04:53:36.125 -> State changing from: 0 to 1
04:53:36.215 -> Setting up AP: ESP32-7936376
04:53:36.215 -> With default password: <hidden>
04:53:36.215 -> AP IP address: 192.168.4.1
04:53:36.215 -> State changed from: 0 to 1

```
Same behaviour with the AP no luck on ESP32-7936376 / ESP32Geiger.
In the 1.15dev version at least the GMC is running. In 1.14.0 no ticks etc.
I've build the GMC during the Session on 08.02.20 in Sindelfingen. No idea which version was flashed.

dl2sba, 2021-03-06:
Reflashed with complete new ESP32 firmware - works now again - sorry for irritation!

closed 2021-03-06 by dl2sba