ecocurious2/MultiGeiger #380: Check / remove sphinx _static warning

M-a-r-i-a-n-n-e, 2021-03-07:
Locally, there is a sphinx warning in case
`html_static_path = ['_static']` in [conf.py](../blob/master/docs/source/conf.py) is used.

The warning:
`WARNING: html_static_path entry '_static' does not exist`

The warning disappears as soon as the line is commented out. However we are not sure if this line is needed for readthedocs.io.

@ThomasWaldmann  fyi, as discussed.

ThomasWaldmann, 2021-03-07:
Tried to reproduce, but I don't get that warning (but some others):
```
user@development:~/w/MultiGeiger/docs$ make html
Running Sphinx v2.3.1
loading pickled environment... failed
failed: Can't get attribute 'Symbol' on <module 'sphinx.domains.c' from '/home/user/.local/lib/python3.5/site-packages/sphinx/domains/c.py'>
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 12 source files that are out of date
updating environment: [new config] 12 added, 0 changed, 0 removed
reading sources... [100%] usage                                                                                                                                                                                                         
/home/user/w/MultiGeiger/docs/source/changes.rst:7: WARNING: Title underline too short.

V1.15.0-dev (not released yet)
------------------
/home/user/w/MultiGeiger/docs/source/setup_lora.rst:120: WARNING: Unexpected indentation.
/home/user/w/MultiGeiger/docs/source/setup_lora.rst:122: WARNING: Block quote ends without a blank line; unexpected unindent.
/home/user/w/MultiGeiger/docs/source/setup_lora.rst:165: WARNING: image file not readable: images/http_integration_filled.png
looking for now-outdated files... none found
pickling environment... done
checking consistency... done
preparing documents... done
writing output... [100%] usage                                                                                                                                                                                                          
generating indices...  genindexdone
writing additional pages...  searchdone
copying images... [100%] ../../images/ble_check02.jpg                                                                                                                                                                                   
copying static files... ... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 4 warnings.

The HTML pages are in build/html.
```

M-a-r-i-a-n-n-e, 2021-03-07:
Interesting. I'm running Sphinx v3.5.1 maybe that's the difference.
I can fix the setup_lora in the next pull request, these are just some minor issues, as far as I can see.

ThomasWaldmann, 2021-03-07:
readthedocs even runs on sphinx 1.8.5, so we need to be careful in case there was any incompatible change.

M-a-r-i-a-n-n-e, 2021-03-07:
Good point. Changed the headline.
Btw: the warnings you've got are now fixed in #382