ecocurious2/MultiGeiger #381: Move gettext translations for readthedocs to the docs/source subfolder

M-a-r-i-a-n-n-e, 2021-03-07:
The [/locales folder](../tree/master/locales) with the gettext translations for readthedocs might be moved to [the /docs/source](../tree/master/docs/source) subfolder to bundle all the sources for readthedocs in the source folder.

As far as I can see, the paths have to be changed in [config.py](../blob/master/docs/source/conf.py) locale_dirs and the [transifex config file](.tx/config).

Maybe also the [/.tx folder](../tree/master/.tx) with the [transifex](https://www.transifex.com/thomaswaldmann/multigeiger) config could be moved into the source folder, too.

@ThomasWaldmann  fyi, as discussed.

ThomasWaldmann, 2021-03-07:
What if we translate other stuff later, that is not in /docs/source? Wouldn't the cause the opposite issue and be even more weird then?

M-a-r-i-a-n-n-e, 2021-03-07:
I would assume that you'll have another folder for translations for e.g. the GUI, yes. Probably in a subfolder of the code repo. But it is just a suggestion, because I was surprised to find the translations outside of the docs folder.