ecocurious2/MultiGeiger #383: Remove already migrated files from repository

M-a-r-i-a-n-n-e, 2021-03-07:
I suggest that already migrated files will be removed from the repository to avoid duplicates.

This might apply, but is not limited, to:
- README [DE](../blob/master/README-de.md) as well as [EN](../blob/master/README-de.md) (have a look first, if everything is now in https://multigeiger.readthedocs.io )
- The images in the [Bilder](../tree/master/docs/Bilder) folder (are they needed any longer? I cannot see any references.)
- The [Bauanleitung](../tree/master/docs/Bauanleitung)

@ThomasWaldmann @WwwBoehriDe fyi

ThomasWaldmann, 2021-03-09:
In general, I PR should not duplicate stuff (except for specific / very good reasons).

So, if you duplicated content in your PR, you should deduplicate it by removing it at the old position.

ThomasWaldmann, 2021-03-21:
The german README needed copy&pasting into transifex - I did that right now. Now everything from the README-*.md is in our *.rst docs, thus I removed the README-*.md from git in PR ...

ThomasWaldmann, 2021-06-18:
shifted to next milestone as i did not see anybody actively working on this.