ecocurious2/MultiGeiger #401: Increase WiFi stability

M-a-r-i-a-n-n-e, 2021-03-22:
No matter how close I put the ESP32 to my router, the connection is not stable, cf the log of approx. half a day:
```
	Line 107: [E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
	Line 137: [E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 113, "Software caused connection abort"
	Line 168: [E][WiFiClient.cpp:232] connect(): connect on fd 58, errno: 118, "Host is unreachable"
	Line 255: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 285: [E][WiFiClient.cpp:258] connect(): socket error on fd 56, errno: 113, "Software caused connection abort"
	Line 315: [E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
	Line 411: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 441: [E][WiFiClient.cpp:258] connect(): socket error on fd 61, errno: 113, "Software caused connection abort"
	Line 471: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 539: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 569: [E][WiFiClient.cpp:258] connect(): socket error on fd 57, errno: 113, "Software caused connection abort"
	Line 599: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 629: [E][WiFiClient.cpp:258] connect(): socket error on fd 57, errno: 113, "Software caused connection abort"
	Line 659: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 727: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 853: [E][WiFiClient.cpp:232] connect(): connect on fd 63, errno: 118, "Host is unreachable"
	Line 969: [E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
	Line 999: [E][WiFiClient.cpp:258] connect(): socket error on fd 60, errno: 113, "Software caused connection abort"
	Line 1029: [E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
```
(I'm using V.1.15.0, but I had the same issues with WiFi before and also tested it over several weeks using a second Multigeiger.)

It would be good to find a way to increase WiFi stability. Maybe there are timeouts due to blocking code or some other issues.

M-a-r-i-a-n-n-e, 2021-03-23:
I took another look in the log: in case of "Host is unreachable", neither Madavi nor sensor.community could be reached.
However, "Software caused connection abort" **always** strikes after sending to Madavi:
```
GEIGER: Sending to Madavi ...
GEIGER: Sent to Madavi, status: ok, http: 200 200
GEIGER: Sending to sensor.community ...
[E][WiFiClient.cpp:258] connect(): socket error on fd 58, errno: 113, "Software caused connection abort"
GEIGER: Error on sending POST: -1
GEIGER: Sent to sensor.community, status: error, http: -1 201
Not connected. Try reconnect...
State changing from: 4 to 3
Connecting to [xxx] (password is hidden)
State changed from: 4 to 3
WiFi connected
IP address: 192.168.178.xxx
State changing from: 3 to 4
Accepting connection
State changed from: 3 to 4
```
Today I will test if running without sending to Madavi solves the problem.

ThomasWaldmann, 2021-03-23:
Notable: one POST is answered with "201 Created" (not with "200 OK" as we might expect).

ThomasWaldmann, 2021-03-25:
Could there be too much interference in the 2.4GHz wifi channel your router is using? Check with a wifi analyzer app or with the builtin tool in the router (if any). Which channel do you use?

rexfue, 2021-03-25:
At the moment I'm testing 3 Geigers: two 'normal' ones and one Stick. One of the normal ones shows a similar effect: it very often loses WLAN and tries to reconnect. The other one is stable. Both are at the same place at my table. Therefore I suspect, there is a hardware problem with the one.

M-a-r-i-a-n-n-e, 2021-03-25:
Well, but then there are many hardware defects, as both of mine are affected.
@ThomasWaldmann I tested in two different locations with two different APs. Each time close to the AP or repeater.

ThomasWaldmann, 2021-03-25:
@M-a-r-i-a-n-n-e another idea: considering the small antenna of the esp32 is of rather low quality: does it get better if you move or rotate the device a few centimeters?

M-a-r-i-a-n-n-e, 2021-03-25:
Unfortunately not.

ThomasWaldmann, 2021-03-25:
Hmm, guess I am out of ideas then. Also not sure this is something we can influence from the firmware (== from our code).

ThomasWaldmann, 2021-03-27:
we can get some wifi fixes via #408.

M-a-r-i-a-n-n-e, 2021-05-15:
Works for me, thank you for fixing!

> GEIGER: Sending to Madavi ...
> GEIGER: Sent to Madavi, status: ok, http: 200 200
> GEIGER: Sending to sensor.community ...
> GEIGER: Sent to sensor.community, status: ok, http: 201 201

closed 2021-05-15 by M-a-r-i-a-n-n-e