ecocurious2/MultiGeiger #404: bme680 / lora warnings

ThomasWaldmann, 2021-03-25:
```
/home/user/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp: In function 'const Arduino_LMIC::HalPinmap_t* Arduino_LMIC::GetPinmap_ThisBoard()':
/home/user/Arduino/libraries/MCCI_LoRaWAN_LMIC_library/src/hal/getpinmap_thisboard.cpp:69:72: note: #pragma message: Board not supported -- use an explicit pinmap
         #pragma message("Board not supported -- use an explicit pinmap")
                                                                        ^
/home/user/Arduino/libraries/Adafruit_BME680_Library/Adafruit_BME680.cpp: In member function 'bool Adafruit_BME680::endReading()':
/home/user/Arduino/libraries/Adafruit_BME680_Library/Adafruit_BME680.cpp:418:20: warning: overflow in implicit constant conversion [-Woverflow]
     gas_resistance = NAN;
```

t-pi, 2021-05-15:
Is the BME680 warning still present with the switch to Adafruit BME680 lib V2.0.0? I didn't observe it here recently.

ThomasWaldmann, 2021-05-15:
Right, I did not see them either when making 1.15.1, closing...

closed 2023-10-15 by ThomasWaldmann