ecocurious2/MultiGeiger #408: try arduino-esp32 1.0.6

ThomasWaldmann, 2021-03-27:
short after the 1.0.5 release (~3w ago), there is now a 1.0.6 release.

if there is no regression, we should do next release based on this.

ThomasWaldmann, 2021-03-27:
@rexfue please update with any insights after trying it out.

ThomasWaldmann, 2021-03-27:
upgraded to 1.0.6 and flashed again:
- boards.txt needed patching again (see `misc/arduino-esp32/...` in our repo) - the patch still applies cleanly
- after that, flashing from arduino ide worked again and multigeiger is ticking normally
- did not do http data uploading tests yet...

rexfue, 2021-03-27:
Don't know what happened, I can't get a stable WLAN connection with the normal Heltec Kit. Only the Wireless Stick connects easily and stable. I need to investigate further - but I cannot continue until later next week.

M-a-r-i-a-n-n-e, 2021-03-28:
@rexfue Same here. Just to be sure, I played around with delay and yield and threw the Lora-related Code out and built it with the original board environment, but it did not help. I also tried it with VSCode as well as with the Arduino IDE.
@ThomasWaldmann why should I patch the boards.txt? As far as I know I just have to configure it again and that's it?

I also tried other versions of espressif32 and they did not work either. Can someone provide an old binary that really worked on the Heltec WiFi Kit 32 v2? (The binary @WwwBoehriDe  flashed did not work very well for me and all newer versions I tried in the previous weeks lead to WiFi connection losses more or less frequently. My settings always were: no sound, local password set, WiFi set, upload to both destinations.)

ThomasWaldmann, 2021-03-28:
@M-a-r-i-a-n-n-e the patching is needed so one can set the partitioning from the arduino IDE's menu, see the .txt in the same directory as the patch (`misc/arduino-esp32/...`). It's required for the **generic** release-binary building.

M-a-r-i-a-n-n-e, 2021-03-29:
@ThomasWaldmann Which of your binaries is 100% sure to work with WiFi on the WiFi Kit?

Regarding the patch: unfortunately, it changes nothing for me. From you manual, I cannot select: 

> Flash size: “4MB(32Mb)”

ThomasWaldmann, 2021-03-29:
@M-a-r-i-a-n-n-e AFAIK, selecting the flash size is not needed as it is autodetected. At least that worked for my 4MB device.

For more infos, please read: https://github.com/ecocurious2/MultiGeiger/blob/master/misc/arduino-esp32/partition-menu-1.0.5.txt

M-a-r-i-a-n-n-e, 2021-03-29:
ok, well, then I have no idea, because your binaries from github also do not lead to a stable WiFi connection (tried it down to 1.13.0 already).

ThomasWaldmann, 2021-04-09:
From last meeting: people have issues with 1.0.6 and wifi, so let's continue with 1.0.5 for now.

ThomasWaldmann, 2021-05-08:
So, 1.15.0 and 1.15.1 releases were / will be made with arduino-esp32 1.0.5.

Later releases should either stay on 1.0.5 or use something > 1.0.6.

Seems like arduino-esp32 devs are working on a 2.0 release (only alphas yet).

M-a-r-i-a-n-n-e, 2021-05-15:
I just tried it again with the current code and until now WiFi works well.

ThomasWaldmann, 2021-05-15:
You mean 1.15.1 release?

M-a-r-i-a-n-n-e, 2021-05-15:
Just the master branch.

M-a-r-i-a-n-n-e, 2021-05-16:
@ThomasWaldmann Out of curiosity I also tested your 1.15.1 release binary. It still fails to connect to WiFi at all, so I do not understand why framework-arduinoespressif32 <3.2.0 (1.0.5) should be better than 3.2.0 (1.0.6). Compiling with previous versions (3.1.1, 3.1.0, 3.0.0, ...) did not work for me, as WiFi was unstable for my two WiFi Kit 32.

Compiling with [framework-arduinoespressif32 3.2.0 (1.0.6)](https://github.com/platformio/platform-espressif32/releases) in VScode works for me, so I can give my Multigeigers another try.

ThomasWaldmann, 2021-05-16:
@M-a-r-i-a-n-n-e 1.0.5 being better than 1.0.6 is what i got from the first few posts in this ticket (including one from you) and also from feedback in a jitsi meeting.

t-pi, 2021-05-16:
Apparently the Arduino BLE stack also did his part in thwarting the stability... Since the move to NimBLE lib, both my MultiGeigers run rock-stable (no reboots anymore, the normally-on is currently at BLE packet counter 18424, i.e. ~51h run time and by my guess it has rolled over once already...)

M-a-r-i-a-n-n-e, 2021-05-17:
@t-pi In March, I even removed all BLE references from the code and as I wrote above, I could compile no version with stable WiFi at all and even the older version that @WwwBoehriDe initially flashed for me had issues with WiFi. So I guess that mainly the new framework-arduinoespressif32 did the trick for me. I'm really glad it works!

ThomasWaldmann, 2021-05-19:
OK, so I guess I'll make a new release soon, with:
- arduino-esp32 1.0.6 release based on IDF release/v3.3.5 85c43024c.
- nimBLE and IotWebConf update from master

I also linked a new patch from the crash-reboot ticket #314, could try to get that in also.

Does that sound good to everybody concerning WiFi stability?

ThomasWaldmann, 2021-05-20:
https://github.com/espressif/arduino-esp32/issues/4980#issuecomment-844736582 < notable.

ThomasWaldmann, 2021-06-03:
https://ecocurious.slack.com/archives/CGBLH04GJ/p1622141846004600 < t.pi: 1.0.6 + nimBLE + local alarm PR uptime >= 7d.

ThomasWaldmann, 2021-08-15:
Building release 1.16.0 with arduino-esp32 1.0.6 now...

closed 2021-08-15 by ThomasWaldmann