ecocurious2/MultiGeiger #409: hardware: dark OLED display

ThomasWaldmann, 2021-03-27:
I have a Multigeiger with the Heltec WiFi Kit 32 microcontroller module since about 1-2y.

I was rather inactive with this project for quite a while and the stuff was in storage at room temperature.

When I got more active again, I noticed that the OLED display is a lot darker than it used to be.
At normal daytime indoors lighting, it is barely readable any more and the left side is a bit darker than the right side.

Any idea what's the root cause for this? Is there a fix?

Other than that, that specific Multigeiger seems to work normally.

rexfue, 2021-03-27:
Was it on or off during storage time? My experience is that the OLEDs slowly break down after 1 to 2 years of operation.

ThomasWaldmann, 2021-03-27:
It was off.

rexfue, 2021-03-28:
Hmm, then I don't know.

ThomasWaldmann, 2021-05-20:
closing this, maybe a defect specific to my device.

closed 2021-05-20 by ThomasWaldmann