ecocurious2/MultiGeiger #410: change log format

ThomasWaldmann, 2021-03-27:
current serial-usb log output format is like:
```
GEIGER: Measured THP: T=25.00 H=29 P=100157
GEIGER:         19           11129   1.707251  0.139036         2     1728   1351734  1.278358  0.104108
```

I guess it would be better to parse if formatted like that:
```
GEIGER: THP:      25.00              29     100157
GEIGER: RAD:         19           11129   1.707251  0.139036         2     1728   1351734  1.278358  0.104108
```

rexfue, 2021-03-27:
I suggest to include a timestamp as well; at least if the time is valid.

GEIGER: 2021-03-27 12:00:23 THP:      25.00              29     100157
GEIGER: 2021-03-27 12:00:23 RAD:         19           11129   1.707251  0.139036         2     1728   1351734  1.278358  0.104108

ThomasWaldmann, 2021-03-27:
Guess the only time-is-valid check we can do there is something like `year > 2020`.

But having an optional timestap component makes parsing more difficult, do we want that?

rexfue, 2021-03-27:
I don't think ist is really difficult ; use a regex (i.e. /GEIGER.*THP/). My demand for the timestamp is: I want to know when the Reboot - if ever -  happens ( or other irreregular events)

ThomasWaldmann, 2021-05-15:
Now looks like this:
```
GEIGER: 1970-01-01T00:00:00 Simple Multi-Geiger, Version V1.16.0-dev
GEIGER: 1970-01-01T00:00:00 -------------------------------------------------------------------------------------------------
GEIGER: 1970-01-01T00:00:00      GMC_counts Time_difference Count_Rate Dose_Rate HV Pulses Accu_GMC Accu_Time Accu_Rate Accu_Dose
GEIGER: 1970-01-01T00:00:00        [Counts]            [ms]      [cps]   [uSv/h]       [-] [Counts]      [ms]     [cps]   [uSv/h]
GEIGER: 1970-01-01T00:00:00 -------------------------------------------------------------------------------------------------
State changing from: 0 to 2
Setting up AP: ESP32-5088740
Use password: <hidden>
AP IP address: 192.168.4.1
AP timeout (ms): 30000
State changed from: 0 to 2
GEIGER: 1970-01-01T00:00:12 RAD:         11           11236   0.978996  0.079728         7       11     11236  0.978996  0.079728
GEIGER: 1970-01-01T00:00:22 RAD:         17            9742   1.745022  0.142112         3       28     20978  1.334732  0.108699
State changing from: 2 to 3
Connecting to [XXX] (password is hidden)
WiFi timeout (ms): 30000
State changed from: 2 to 3
WiFi connected
IP address: x.x.x.x
State changing from: 3 to 4
Accepting connection
State changed from: 3 to 4
GEIGER: 2021-05-15T15:23:51 RAD:          8           10825   0.739030  0.060186         1       36     31803  1.131969  0.092186
GEIGER: 2021-05-15T15:24:01 RAD:          8            9243   0.865520  0.070487         1       44     41046  1.071968  0.087300
GEIGER: 2021-05-15T15:24:11 RAD:         11           10831   1.015603  0.082709         1       55     51877  1.060200  0.086341
GEIGER: 2021-05-15T15:24:21 RAD:         14            9738   1.437667  0.117081         1       69     61615  1.119857  0.091200
GEIGER: 2021-05-15T15:24:31 RAD:         20           10160   1.968504  0.160312         1       89     71775  1.239986  0.100983
GEIGER: 2021-05-15T15:24:41 RAD:         11            8592   1.280261  0.104263         1      100     80367  1.244292  0.101333
GEIGER: 2021-05-15T15:24:51 RAD:         15           11052   1.357220  0.110530         1      115     91419  1.257944  0.102445
GEIGER: 2021-05-15T15:25:01 RAD:          6            9772   0.613999  0.050003         1      121    101191  1.195758  0.097381
GEIGER: 2021-05-15T15:25:11 RAD:         21           10075   2.084367  0.169748         1      142    111266  1.276221  0.103934
GEIGER: 2021-05-15T15:25:21 RAD:         14           10310   1.357905  0.110586         2      156    121576  1.283148  0.104498
GEIGER: 2021-05-15T15:25:31 RAD:         11           10273   1.070768  0.087202         1      167    131849  1.266600  0.103150
GEIGER: 2021-05-15T15:25:41 RAD:         17            8473   2.006373  0.163396         2      184    140322  1.311270  0.106788
GEIGER: 2021-05-15T15:25:50 THP: 24.48 36 97971
GEIGER: 2021-05-15T15:25:51 RAD:         13           10819   1.201590  0.097856         2      197    151141  1.303419  0.106148
GEIGER: 2021-05-15T15:26:01 RAD:         11            9903   1.110775  0.090460         1      208    161044  1.291572  0.105184
GEIGER: 2021-05-15T15:26:11 RAD:          9           10517   0.855757  0.069692         1      217    171561  1.264856  0.103008
GEIGER: 2021-05-15T15:26:21 RAD:         14            9380   1.492537  0.121550         2      231    180941  1.276659  0.103969
GEIGER: 2021-05-15T15:26:31 RAD:         12            9859   1.217162  0.099124         2      243    190800  1.273585  0.103719
GEIGER: 2021-05-15T15:26:41 RAD:         10            9401   1.063717  0.086628         2      253    200201  1.263730  0.102916
```

ThomasWaldmann, 2021-05-15:
We could make it even prettier (and maybe get rid of RAD: and THP: again) if we just displayed t, h, p at the end of every line (not necessarily meaning it really updates all 10s).

ThomasWaldmann, 2021-05-20:
decision from meeting: add 3 columns for THP.

PR #430 (2021-06-18): improve data logging
- more formatting consistency
- add timestamp
- add t/h/p to rightmost columns
- repeat header output for better readability

Fixes #410.
closed 2021-06-18 by ThomasWaldmann