ecocurious2/MultiGeiger #416: Add simple buzzer alarm based on historic and current values

t-pi, 2021-05-04:
We've discussed this during the meeting already.
One option is an email via a script checking the online data - not fully in focus here.
Another option is a buzzer alarm, if some thresholds are reached, e.g.

1. long term averaged value > 500nSv/h?
2. (short term values over e.g. 1 min) > (long term values * 3)

With the interrupt based approach from a recent version,  @ThomasWaldmann, could we play an alarm sound while keeping the system running? In that case options (1) and (2) could be easily integrated into the firmware, e.g. as #define options.

Thomas, @rxf, @WwwBoehriDe, what do you think about the thresholds?

ThomasWaldmann, 2021-05-05:
The code has a little sequencer-like tone playing system, so technically i don't see a problem (you just won't hear ticks and tones at the same time).

But **if** a warning functionality is wanted, something server based / based on the online data would be more useful. I don't think we want to send email or other notifications directly from the multigeiger.

Also, a buzzer alarm is only useful if you are close to the device, which is usually not the case for the current outside deployment method.

BTW, there has already been some discussion somewhere whether one wants to send out "alarm notifications", especially if that functionality is rather public. That could be useful but it could also cause a panic just because of some malfunction.

t-pi, 2021-05-08:
Indeed, my local code works with the sequencer, after a few small modifications. Did add this on top of my NimBLE PR, so will push this after merge.

Regarding the local alarm being of no use, I disagree. 

We don't see radiation, and after a few days I rarely look up the values online. Of course a mail alarm will help, and if IFTTT works as discussed last Thursday, this can also be implemented quickly without the need to own a server to have the mail script running. However my MultiGeiger sits in my flat, so an alarm would be much more immediate. The one at my parents' garden would also be heard while staying (or passing) outside the house - and probably all Geigers are sitting in a few meters distance from a wall plug, so they should be heard, when in proximity.

So I implemented it as a firmware compilation option as of now. Whoever wants this, must thus activate the compile option in ```userdefines.h``` and can alter the value (0.5µSv/h accumulated or 3x accumulated). The alarm sequence will play every 10s, when ```publish``` is called. You can then try it out and listen for yourself if this could work - after merging NimBLE 😄

ThomasWaldmann, 2021-05-08:
OK, it it is useful to you, go on. I just imagined my usecase (when deployed to garden, i usually won't hear it - maybe with a delay of days which would likely mean that i get something from the internet news first).

No compile option please, that is not useful for most people who just use the binary from github releases.

So it needs some configurable options like enabled alarms (currently just "buzzer", but could be LED or http request also) and an alarm threshold value.

t-pi, 2021-05-09:
Okay, will further add the IoTWebConf modifications to be able to set and change via web interface.

t-pi, 2021-07-06:
Adressed by #427

closed 2021-07-06 by t-pi