ecocurious2/MultiGeiger #431: support tools -> serial plotter from arduino IDE?

ThomasWaldmann, 2021-05-15:
it's quite simple, it just needs space or tab separated values printed to serial-usb.

as the vertical scaling automatically adjusts and there is only **one** vertical scale, showing values that display nicely within that range would make a pretty display.

value candidates:
- radiation in uSv/h (or nSv/h)
- raw GM counts / measurement interval
- temperature in deg celsius
- humidity in %

maybe rather not:
- pressure (mbar == hectopascal is usually about 1000)
- anything "accumulated" also does not make much sense

rexfue, 2021-05-15:
Nice idea. How to decide what to send? And anyway, how to distinguish between logging and sending data? Maybe user defines.h or vie web (iotwebconfig) ?

ThomasWaldmann, 2021-05-15:
Yeah, there are some issues:

- misc. output of esp32 core not related to our data (e.g. wifi messages)
- because of these, one needs to start serial plotter late (when only plot data is coming)
- didn't manage to display labels in a controlled way
- only hacks to influence vertical scale

It's in my logging PR, see there.

ThomasWaldmann, 2021-05-20:
Guess I'll drop this again, due to these issues.

closed 2021-05-20 by ThomasWaldmann