ecocurious2/MultiGeiger #437: TTN v3: parameters and en-/decoder

Marwe, 2021-07-04:
I want to migrate my LoRA applications from v2 (EOL 2021-12) to v3, does anyone have experience with it or already wrote some updated code snippet?

As far as I can see, the application definition needs more parameters (LoRA specific, e.g. versions, which I did not really know) and the js Decoder function needs adaption.
May not be too difficult, though I am not really a js coder. I can see some patterns looking at 
the [payload formater examples](https://www.thethingsindustries.com/docs/integrations/payload-formatters/javascript/), but find it too difficult to transpile due to lacking background.

There is only one variable `input` with `input.bytes` and `input.fPort`, the rest of the function may be similar. I don't yet get the `events` and where/how they are transferred (port definition? `data.event = events[input.fPort]`).

I don't know if a downlink function would be necessary, it may be needed (with a reversed logic).

There is a [device repository](https://www.thethingsindustries.com/docs/integrations/payload-formatters/device-repo/) with payload formatters, maybe this is the right place for this kind of code?

rexfue, 2021-07-05:
Migration from V2 to V3 for devices is easy: follow the description at https://www.thethingsnetwork.org/docs/the-things-stack/migrate-to-v3/migrate-using-console/.
For the payload decoder there is a 'Helper Script'written by Severin Schols, that lets you use your old V2 payload formatter, see: https://www.thethingsnetwork.org/community/munich/post/using-ttn-v2-payload-decode-functions-with-ttn-v3
I tried it and it works.

sbiermann, 2021-07-19:
I can confirm, it works like a charm after following the two descriptions. I'm now using old 1.12.0 version with TTN V3, no need for a new flashing.